BCDA-APS/usaxs-bits #6: Support the 12ID monochromator with bluesky

prjemian, 2024-04-19:


prjemian, 2024-04-19:
Started asking in BCDA for experience with this monochromator.

prjemian, 2024-04-19:
EPICS database: /net/s12dserv/xorApps/epics/synApps_6_1/ioc/12ida1/db/SB_DCM.db

IOC: https://git.aps.anl.gov/12ID/12ida1

prjemian, 2024-04-19:
Review the existing SPEC controls.

prjemian, 2024-04-19:
In the EPICS database...

Choices like this need to be considered by the users of the controls:

```
record(bo, "$(P)Z2_Mode") {  field(ZNAM, "Freeze Z2")  field(ONAM, "Move Z2")}
```

Bluesky can avoid exposing this PV.

prjemian, 2024-04-19:
Thinking energy and wavelength should both be positioner Components so they act like motors, as part of a monochromator Device.  Will need additional Components for things like `Z2_Mode` PV.  Also, if there is any caution of _adjust this before that_, we'll need Components for each such PV and a procedure for such adjustment.

jilavsky, 2024-10-25:
Tested 10-26 and looks like we need tolerance value. Trying 0.0002

jilavsky, 2024-10-25:
Success, added tolerance and get mono E scans, however, the BS crashed again, see issue #7. Looks like #7 is showstopper at this time. Am I the only one or why is this the case?

jilavsky, 2024-10-28:
Monochromator works as needed now, energy can be scanned, energy and wavelength can be recorded to metadata. This issue is fixed.

closed 2024-10-28 by jilavsky