BCDA-APS/usaxs-bits #7: Crashed BlueSky

jilavsky, 2024-10-25:
Was trying to run scan on monochromator, but it was hanging likely due to tolerance issue. Didctrl-C. 2x and RE.abort. After this run scan on s_stage.x which worked fine, but after that looks like BS crashed:

```
In [10]: RE(bp.scan([scaler0],s_stage.x,-120,-122,21))

Transient Scan ID: 8     Time: 2024-10-25 09:06:42
Persistent Unique Scan ID: 'fc998cc0-a063-4b86-9d58-176a372e2849'
New stream: 'baseline'
New stream: 'primary'                                                                                                                                                                
/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/callbacks/best_effort.py:208: UserWarning: Starting a Matplotlib GUI outside of the main thread will likely fail.
  fig = self._fig_factory(fig_name)
+-----------+------------+------------+------------+------------+
|   seq_num |       time |  s_stage_x |   I0_USAXS |   PD_USAXS |
+-----------+------------+------------+------------+------------+
|         1 | 09:06:46.7 | -119.999992 |       3795 |      57194 |
|         2 | 09:06:48.0 | -120.100013 |       3796 |      57749 |                                                                                                                   
|         3 | 09:06:49.3 | -120.199946 |       3795 |      57249 |                                                                                                                   
|         4 | 09:06:50.6 | -120.299989 |       3796 |      59234 |                                                                                                                   
|         5 | 09:06:51.8 | -120.399949 |       3796 |      57167 |                                                                                                                   
|         6 | 09:06:53.1 | -120.499975 |       3796 |      59530 |                                                                                                                   
|         7 | 09:06:54.4 | -120.599981 |       3796 |      58353 |                                                                                                                   
|         8 | 09:06:55.6 | -120.699963 |       3795 |      57001 |                                                                                                                   
|         9 | 09:06:56.9 | -120.799963 |       3795 |      59245 |                                                                                                                   
|        10 | 09:06:58.2 | -120.899971 |       3795 |      59551 |                                                                                                                   
|        11 | 09:06:59.5 | -120.999974 |       3795 |      58555 |                                                                                                                   
|        12 | 09:07:00.7 | -121.099958 |       3796 |      58055 |                                                                                                                   
|        13 | 09:07:02.0 | -121.199999 |       3795 |      56445 |                                                                                                                   
|        14 | 09:07:03.3 | -121.299980 |       3796 |      57453 |                                                                                                                   
|        15 | 09:07:04.6 | -121.400005 |       3795 |      57362 |                                                                                                                   
|        16 | 09:07:05.9 | -121.500000 |       3796 |      57060 |                                                                                                                   
|        17 | 09:07:07.2 | -121.600000 |       3796 |      56959 |                                                                                                                   
|        18 | 09:07:08.5 | -121.699979 |       3798 |      57164 |                                                                                                                   
|        19 | 09:07:09.8 | -121.800000 |       3796 |      56457 |                                                                                                                   
|        20 | 09:07:11.1 | -121.899986 |       3796 |      56576 |                                                                                                                   
|        21 | 09:07:12.3 | -121.999986 |       3796 |      56656 |                                                                                                                   
+-----------+------------+------------+------------+------------+
generator scan ['fc998cc0'] (scan num: 8)
Out[10]: ('fc998cc0-a063-4b86-9d58-176a372e2849',)

In [11]: Traceback (most recent call last):
  File "/share1/bluesky/conda_envs/bluesky_2024_3/bin/ipython", line 8, in <module>
    sys.exit(start_ipython())
             ^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/__init__.py", line 130, in start_ipython
    return launch_new_instance(argv=argv, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/traitlets/config/application.py", line 1075, in launch_instance
    app.start()
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/terminal/ipapp.py", line 317, in start
    self.shell.mainloop()
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/terminal/interactiveshell.py", line 917, in mainloop
    self.interact()
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/terminal/interactiveshell.py", line 902, in interact
    code = self.prompt_for_code()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/terminal/interactiveshell.py", line 845, in prompt_for_code
    text = self.pt_app.prompt(
           ^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/prompt_toolkit/shortcuts/prompt.py", line 1035, in prompt
    return self.app.run(
           ^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/prompt_toolkit/application/application.py", line 978, in run
    result = loop.run_until_complete(coro)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/asyncio/base_events.py", line 641, in run_until_complete
    self.run_forever()
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/asyncio/base_events.py", line 608, in run_forever
    self._run_once()
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/asyncio/base_events.py", line 1898, in _run_once
    event_list = self._selector.select(timeout)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/prompt_toolkit/eventloop/inputhook.py", line 150, in select
    self.inputhook(InputHookContext(self._r, input_is_ready))
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/terminal/pt_inputhooks/tk.py", line 88, in inputhook
    wait_using_filehandler()
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/terminal/pt_inputhooks/tk.py", line 65, in wait_using_filehandler
    root.createfilehandler(inputhook_context.fileno(), _tkinter.READABLE, done)
RuntimeError: Calling Tcl from different apartment

If you suspect this is an IPython 8.27.0 bug, please report it at:
    https://github.com/ipython/ipython/issues
or send an email to the mailing list at ipython-dev@python.org

You can print a more detailed traceback right now with "%tb", or use "%debug"
to interactively debug it.

Extra-detailed tracebacks for bug-reporting purposes can be enabled via:
    %config Application.verbose_crash=True

Exception ignored in atexit callback: <bound method Gcf.destroy_all of <class 'matplotlib._pylab_helpers.Gcf'>>
Traceback (most recent call last):
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/matplotlib/_pylab_helpers.py", line 81, in destroy_all
    manager.destroy()
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/matplotlib/backends/_backend_tk.py", line 562, in destroy
    self.canvas._tkcanvas.after_cancel(self.canvas._idle_draw_id)
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/tkinter/__init__.py", line 893, in after_cancel
    data = self.tk.call('after', 'info', id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: main thread is not in main loop
(/share1/bluesky/conda_envs/bluesky_2024_3) bash-4.4$ 

```

jilavsky, 2024-10-25:
When I tried to continue with command line, it seemed unresponsive and when I hit enter, it opened GUI with DataBroker catalog, when I tried to close it it stated it was not saved. Closed anyway, but looks like DataBroker was active and waiting for something.

prjemian, 2024-10-25:
Problem here is that matplotlib is being configured with Tk. That's the
problem to be fixed.

jilavsky, 2024-10-25:
Matplotlib seems to be imported in generic configuration of bluesky, for notebook as well as console session.

prjemian, 2024-10-25:
Seen recently and entirely clear why.  Here is the clear (to me) identification of this situation:
[image]

Next, I'm going to question the steps you executed to create this conda environment.  (There might be something to change in the `environment.yml` file.)

prjemian, 2024-10-25:
This thread from Bluesky's Mattermost discussion:
[image]

The [supported backends](https://github.com/bluesky/bluesky/blob/924d10c4d3494fef083e586400133924191f6df2/src/bluesky/utils/__init__.py#L975-L977) (at this time) are: `"nbAgg"`, `"Qt4Agg"`, and `"Qt5Agg"`.

I'm thinking that the conda environment did not get a requirement to install one of the Qt packages.  If this is left to `pip`, errors like this one will occur.

prjemian, 2024-10-25:
Related to: https://github.com/prjemian/model_instrument/pull/5

jilavsky, 2024-10-25:
Short version : recreating bluesky_2024_3 from yml file does not help.
Long version: deleted bluesky_2024_3 environment, deleted yml file for 2024_3, downloaded yml file per instructions, created new environment bluesky_2024_3. Started BS, setup scaler and run a scan. Same problem, BS crashes after scan finishes.
This is something of warning one gets ahead of scan itself
'''
Transient Scan ID: 10     Time: 2024-10-25 13:43:19
Persistent Unique Scan ID: '53091dbd-3f7c-48a5-ab9a-8dafdc31ca54'
New stream: 'baseline'
New stream: 'primary'                                                                                                                                                              
/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/callbacks/best_effort.py:211: UserWarning: Starting a Matplotlib GUI outside of the main thread will likely fail.
  fig = self._fig_factory(fig_name)
'''  
Ready to try something else.

prjemian, 2024-10-27:
I'm reinstalling from https://raw.githubusercontent.com/BCDA-APS/bluesky_training/refs/heads/main/bluesky/environments/environment_2024_3.yml (downloading into `~/bluesky/environments/`).

prjemian, 2024-10-27:
With re-installed environment, restarted session.  This is the test:

```py
In [1]: import matplotlib as mpl

In [2]: mpl.backends.backend
Out[2]: 'tkagg'
```

Should be `qtagg`.  Something's off with the environment or the startup.

prjemian, 2024-10-27:
Similar versions with that issue's solved version:

```bash
In [3]: !conda list qt
/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/core/interactiveshell.py:2673: UserWarning: You executed the system command !conda which may not work as expected. Try the IPython magic %conda instead.
  warnings.warn(
# packages in environment at /share1/bluesky/conda_envs/bluesky_2024_3:
#
# Name                    Version                   Build  Channel
pyqt                      5.15.9          py311hf0fb5b6_5    conda-forge
pyqt5-sip                 12.12.2         py311hb755f60_5    conda-forge
pyqtgraph                 0.12.4                   pypi_0    pypi
qt                        5.15.8               hf11cfaa_0    conda-forge
qt-main                   5.15.8              h374914d_26    conda-forge
qt-webengine              5.15.8               h8f589be_8    conda-forge
qtconsole                 5.6.0                    pypi_0    pypi
qtpy                      2.4.1                    pypi_0    pypi
sphinxcontrib-qthelp      2.0.0                    pypi_0    pypi
```

prjemian, 2024-10-27:
At this point: https://github.com/APS-USAXS/bluesky/blob/2e9ba47590f7055ec25af24d327501eb4ba76a54/instrument/mpl/console.py#L17

The backend is `"tkagg"`.  Digging further since this code has not changed ...

prjemian, 2024-10-27:
Need more diagnostics, increasing the xmode level to `"Verbose"`

<detail>

```py
(/share1/bluesky/conda_envs/bluesky_2024_3) usaxs@usaxscontrol ~/bluesky $ blueskyStarter.sh 
==> CONDA_PREFIX=/share1/bluesky/conda_envs/bluesky_2024_3
/share1/bluesky/conda_envs/bluesky_2024_3/bin/python -- Python 3.11.10
Python 3.11.10 | packaged by conda-forge | (main, Oct 16 2024, 01:27:36) [GCC 13.3.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.29.0 -- An enhanced Interactive Python. Type '?' for help.

IPython profile: bluesky
Exception reporting mode: Minimal
/home/beams11/USAXS/bluesky/instrument/_iconfig.py
Activating auto-logging. Current session state plus future input saved.
Filename       : /home/beams11/USAXS/bluesky/.logs/ipython_console.log
Mode           : rotate
Output logging : True
Raw input log  : False
Timestamping   : True
State          : active
I Sun-17:24:43 - ############################################################ startup
I Sun-17:24:43 - logging started
I Sun-17:24:43 - logging level = 10
I Sun-17:24:43 - /home/beams11/USAXS/bluesky/instrument/session_logs.py
I Sun-17:24:43 - /home/beams11/USAXS/bluesky/instrument/collection.py
I Sun-17:24:43 - CONDA_PREFIX = /share1/bluesky/conda_envs/bluesky_2024_3
Exception reporting mode: Verbose
I Sun-17:24:43 - xmode exception level: 'Verbose'
I Sun-17:24:43 - /home/beams11/USAXS/bluesky/instrument/mpl/console.py
PyQt5.__file__='/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/PyQt5/__init__.py'
matplotlib.backends.backend='tkagg'
[TerminalIPythonApp] WARNING | Unknown error in handling startup files:
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
File /share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/core/shellapp.py:403, in InteractiveShellApp._exec_file(self=<IPython.terminal.ipapp.TerminalIPythonApp object>, fname='/home/beams/USAXS/.ipython-bluesky/profile_bluesky/startup/__start_instrument__.py', shell_futures=False)
    399                 self.shell.safe_execfile_ipy(full_filename,
    400                                              shell_futures=shell_futures)
    401             else:
    402                 # default to python, even without extension
--> 403                 self.shell.safe_execfile(full_filename,
        full_filename = '/home/beams/USAXS/.ipython-bluesky/profile_bluesky/startup/__start_instrument__.py'
        self = <IPython.terminal.ipapp.TerminalIPythonApp object at 0x7f7bf35a6990>
        shell_futures = False
    404                                          self.shell.user_ns,
    405                                          shell_futures=shell_futures,
    406                                          raise_exceptions=True)
    407 finally:
    408     sys.argv = save_argv

File /share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/core/interactiveshell.py:2932, in InteractiveShell.safe_execfile(self=<IPython.terminal.interactiveshell.TerminalInteractiveShell object>, fname=PosixPath('/home/beams11/USAXS/.ipython-bluesky/profile_bluesky/startup/__start_instrument__.py'), exit_ignore=False, raise_exceptions=True, shell_futures=False, *where=({'In': [''], 'Out': {}, '_': '', '__': '', '___': '', '__builtin__': <module 'builtins' (built-in)>, '__builtins__': <module 'builtins' (built-in)>, '__doc__': '\nstart bluesky in IPython session\n', '__loader__': None, '__name__': '__main__', ...},))
   2930 try:
   2931     glob, loc = (where + (None, ))[:2]
-> 2932     py3compat.execfile(
        py3compat = <module 'IPython.utils.py3compat' from '/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/IPython/utils/py3compat.py'>
        fname = PosixPath('/home/beams11/USAXS/.ipython-bluesky/profile_bluesky/startup/__start_instrument__.py')
        glob = {'__name__': '__main__', '__doc__': '\nstart bluesky in IPython session\n', '__package__': None, '__loader__': None, '__spec__': None, '__builtin__': <module 'builtins' (built-in)>, '__builtins__': <module 'builtins' (built-in)>, '_ih': [''], '_oh': {}, '_dh': [PosixPath('/home/beams11/USAXS/bluesky')], 'In': [''], 'Out': {}, 'get_ipython': <bound method InteractiveShell.get_ipython of <IPython.terminal.interactiveshell.TerminalInteractiveShell object at 0x7f7bf125af10>>, 'exit': <IPython.core.autocall.ExitAutocall object at 0x7f7bf1143dd0>, 'quit': <IPython.core.autocall.ExitAutocall object at 0x7f7bf1143dd0>, 'open': <function open at 0x7f7bf1b85260>, '_': '', '__': '', '___': '', 'faulthandler': <module 'faulthandler' (built-in)>, 'pathlib': <module 'pathlib' from '/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/pathlib.py'>, 'sys': <module 'sys' (built-in)>, 'path': PosixPath('/home/beams/USAXS/bluesky')}
        loc = None
        self = <IPython.terminal.interactiveshell.TerminalInteractiveShell object at 0x7f7bf125af10>
        shell_futures = False
        self.compile = <IPython.core.compilerop.CachingCompiler object at 0x7f7bf1335e90>
   2933         fname, glob, loc,
   2934         self.compile if shell_futures else None)
   2935 except SystemExit as status:
   2936     # If the call was made with 0 or None exit status (sys.exit(0)
   2937     # or sys.exit() ), don't bother showing a traceback, as both of
   (...)
   2943     # For other exit status, we show the exception unless
[... cut ...]
        _candidates = [(<function _setup_pyqt5plus at 0x7f7b66dcd3a0>, 'PyQt6'), (<function _setup_pyqt5plus at 0x7f7b66dcd3a0>, 'PySide6'), (<function _setup_pyqt5plus at 0x7f7b66dcd3a0>, 'PyQt5'), (<function _setup_pyqt5plus at 0x7f7b66dcd3a0>, 'PySide2')]
    131             "Failed to import any of the following Qt binding modules: {}"
    132             .format(", ".join([QT_API for _, QT_API in _candidates]))
    133         )
    134 else:  # We should not get there.
    135     raise AssertionError(f"Unexpected QT_API: {QT_API}")

ImportError: Failed to import any of the following Qt binding modules: PyQt6, PySide6, PyQt5, PySide2

```

</detail>

prjemian, 2024-10-27:
A clue comes from another [discussion](https://github.com/matplotlib/matplotlib/issues/25673#issuecomment-1506278627).  Testing with this console command: `%gui qt` raise this exception:

```py
In [3]: %gui qt
ERROR:root:/lib64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/PyQt5/../../../libQt5Core.so.5)
```

Is the library file `libQt5Core.so.5` installed here _incompatible_ with libstdc++.so.6 installed by the OS?

prjemian, 2024-10-27:
That library is already installed under `$CONDA_PREFIX/lib/libstdc++.so`:

```bash
(/share1/bluesky/conda_envs/bluesky_2024_3) usaxs@usaxscontrol ~/bluesky $ strings $CONDA_PREFIX/lib/libstdc++.so | grep GLIBCXX | grep 26
GLIBCXX_3.4.26
_ZNSbIwSt11char_traitsIwESaIwEE4_Rep26_M_set_length_and_sharableEm@@GLIBCXX_3.4.5
_ZNSs4_Rep26_M_set_length_and_sharableEm@GLIBCXX_3.4
_ZNSbIwSt11char_traitsIwESaIwEE4_Rep26_M_set_length_and_sharableEm@GLIBCXX_3.4
_ZNSs4_Rep26_M_set_length_and_sharableEm@@GLIBCXX_3.4.5
GLIBCXX_3.4.26
```

## Solution to this problem

Define in bash shell: `export LD_LIBRARY_PATH="${CONDA_PREFIX}/lib".  Added that to the starter script `blueskyStarter.sh`.

prjemian, 2024-10-27:
Test proper operation using simulator:

Start console session as usual: `blueskyStarter.sh`

Load ophyd simulator and scan it:

```py
In [1]: from ophyd.sim import noisy_det, motor

In [2]: RE(bp.scan([noisy_det], motor, -3, 3, 31))

Transient Scan ID: 14     Time: 2024-10-27 18:21:44
Persistent Unique Scan ID: '3236f8b2-93d1-4d55-8f89-f91f29b55b02'
New stream: 'baseline'
New stream: 'primary'
+-----------+------------+------------+------------+
|   seq_num |       time |      motor |  noisy_det |
+-----------+------------+------------+------------+
|         1 | 18:21:44.3 |     -3.000 |      0.009 |
|         2 | 18:21:45.3 |     -2.800 |      0.022 |
|         3 | 18:21:45.3 |     -2.600 |      0.085 |
|         4 | 18:21:45.3 |     -2.400 |      0.105 |
|         5 | 18:21:45.3 |     -2.200 |      0.149 |
|         6 | 18:21:45.3 |     -2.000 |      0.186 |
|         7 | 18:21:45.3 |     -1.800 |      0.267 |
|         8 | 18:21:45.3 |     -1.600 |      0.335 |
|         9 | 18:21:45.3 |     -1.400 |      0.410 |
|        10 | 18:21:45.3 |     -1.200 |      0.405 |
|        11 | 18:21:45.3 |     -1.000 |      0.629 |
|        12 | 18:21:45.3 |     -0.800 |      0.704 |
|        13 | 18:21:45.4 |     -0.600 |      0.867 |
|        14 | 18:21:45.4 |     -0.400 |      0.843 |
|        15 | 18:21:45.4 |     -0.200 |      0.983 |
|        16 | 18:21:45.4 |      0.000 |      0.999 |
|        17 | 18:21:45.4 |      0.200 |      0.911 |
|        18 | 18:21:45.4 |      0.400 |      0.958 |
|        19 | 18:21:45.4 |      0.600 |      0.804 |
|        20 | 18:21:45.4 |      0.800 |      0.751 |
|        21 | 18:21:45.4 |      1.000 |      0.663 |
|        22 | 18:21:45.4 |      1.200 |      0.445 |
|        23 | 18:21:45.4 |      1.400 |      0.468 |
|        24 | 18:21:45.4 |      1.600 |      0.374 |
|        25 | 18:21:45.4 |      1.800 |      0.295 |
|        26 | 18:21:45.4 |      2.000 |      0.218 |
|        27 | 18:21:45.5 |      2.200 |      0.042 |
|        28 | 18:21:45.5 |      2.400 |      0.150 |
|        29 | 18:21:45.5 |      2.600 |     -0.009 |
|        30 | 18:21:45.5 |      2.800 |      0.074 |
|        31 | 18:21:45.5 |      3.000 |     -0.076 |
+-----------+------------+------------+------------+
generator scan ['3236f8b2'] (scan num: 14)
Out[2]: ('3236f8b2-93d1-4d55-8f89-f91f29b55b02',)

```

MatPlotLib/Qt screen from BestEffortCallback:
[image]

prjemian, 2024-10-27:
I'll try to explore why this library was not found by default.  It should have been found with no additional help.  Not obvious why.

jilavsky, 2024-10-28:
This issue is fixed, scan verified to be working as expected. Thank you!

closed 2024-10-28 by jilavsky