BCDA-APS/usaxs-bits #10: Filter device = help needed

jilavsky, 2024-11-14:
I am creating my Filter device for Max's ADR-300 PyFilter device. This is 12 filter device, we actually use only 4 Al and 4 Ti positions, the others are empty. This is configured on ioc, so from use point of view this is simply device which takes order value in increasing attenuation which we will store in epics. So instrument will insert 0 (pull all out) or arbitrary value (e.g. 8) which is set by staff for various conditions. 
I think I have very baseline support needed by USAXS, which really needs to read attenuation and set filters per epics stored value.  There is simply order in which combinations of filters are inserted 0 - very large number. 
Support in plans/filters.py needs updating, after this is approved to make some sense... 

Code is in devices/filters.py

Only two values are really used : 
fPos   - 12idPyFilter:FL1:sortedIndex -- is input/output and 
attenuation    - 12idPyFilter:FL1:attenuation_actual -- is currently calculated attenuation, RO. 

I followed the PF4 filter example. 

```
class FilterBank(Device):
    """
    A single module of filters (12-blades).

    .. index:: Ophyd Device; FilterBank

    EXAMPLES::

        pfilter = FilterBank("ioc:", name="pfilter", bank="FL1")

    :see: 
    """

    fPos = FormattedComponent(EpicsSignal, "{prefix}{bank}:sortedIndex", kind="config")
    attenuation = FormattedComponent(EpicsSignalRO, "{prefix}{bank}:attenuation_actual", kind="config")

    def __init__(self, prefix, bank=None, **kwargs):
        self._bank = bank
        super().__init__(prefix, **kwargs)

Filter_AlTi = FilterBank("12idPyFilter:", name="Filter_AlTi",bank="FL1")

```

prjemian, 2024-11-15:
This looks like it should work.  Very similar to [`apstools.devices.Pf4FilterBank`](https://github.com/BCDA-APS/apstools/blob/2013e3bd4855035edc822ac488166e43b9a6b350/apstools/devices/xia_pf4.py#L22).  I presume the ADR-300 PyFilter EPICS support does not have the additional PVs that would allow you to subclass directly from [`apstools.devices.Pf4FilterBank`](https://github.com/BCDA-APS/apstools/blob/2013e3bd4855035edc822ac488166e43b9a6b350/apstools/devices/xia_pf4.py#L22).

prjemian, 2024-11-15:
The only point in adding the additional signal components (PVs) is to report them in metadata.

jilavsky, 2024-11-15:
Does not compile, I get error on "bank" : 
[image]
Not sure where in the file bank offends anything

jilavsky, 2024-11-15:
Fixed it, it needs to be _bank as that is how it is initialized as self._bank... 
```
    fPos = FormattedComponent(EpicsSignal, "{prefix}{_bank}:sortedIndex", kind="config")
    attenuation = FormattedComponent(EpicsSignalRO, "{prefix}{_bank}:attenuation_actual", kind="config")

    def __init__(self, prefix, bank=None, **kwargs):
        self._bank = bank
        super().__init__(prefix, **kwargs)
```

prjemian, 2024-11-15:
Crank up the [verbosity](https://jakevdp.github.io/PythonDataScienceHandbook/01.06-errors-and-debugging.html) on the exception traces for this, to see more detail.  https://github.com/APS-USAXS/bluesky/blob/b8c6971fc5277b3a0f602041bdf0d2c8266d9604/instrument/iconfig.yml#L61

For now, choose `Verbose` until this problem is solved, then set it back.

```yaml
# XMODE_DEBUG_LEVEL: Minimal
XMODE_DEBUG_LEVEL: Verbose
```

jilavsky, 2024-11-15:
Done, I am adding more packages in which I forgot to add before. Nexus, specwriter etc. I will submit new request when I run into issues. This is simple debugging/fixing. =Thanks!