BCDA-APS/usaxs-bits #11: Add transmission to FIlters

jilavsky, 2024-11-16:
This filter box ioc only provides attenuation, humans liketransmission (1/att). 
This code (derivedSignal) doe snto work:

```
class FilterBank(Device):
    """
    A single module of filters (12-blades).

    .. index:: Ophyd Device; FilterBank

    EXAMPLES::

        pfilter = FilterBank("ioc:", name="pfilter", bank="FL1")

    :see: 
    """
 
    class TransmDerivedSignal(DerivedSignal):
        def forward(self, value):
            return 1/value      
         
        def inverse(self, value):
            return 1/value
        

    fPos = FormattedComponent(EpicsSignal, "{prefix}{_bank}:sortedIndex", kind="config")
    attenuation = FormattedComponent(EpicsSignalRO, "{prefix}{_bank}:attenuation_actual", kind="config")
    transmission = TransmDerivedSignal(derived_from=attenuation, name="transmission", metadata=None)

    def __init__(self, prefix, bank=None, **kwargs):
        self._bank = bank
        super().__init__(prefix, **kwargs)

```

jilavsky, 2024-11-16:
Following example here: https://github.com/bluesky/ophyd/blob/ee7ab9ba92ef415e84e315f33191c627bb29c17c/ophyd/tests/test_signal.py#L754

jilavsky, 2024-11-16:
Changed as instructed to:

    transmission = Component(TransmDerivedSignal,derived_from=attenuation, name="transmission", metadata=None)

Still get error:

File /share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/ophyd/device.py:327, in Component.__get__(self=Component(TransmDerivedSignal, derived_from=Form...ame='transmission', metadata=None, kind='normal'), instance=<class 'instrument.devices.filters.FilterBank'> instance, owner=<class 'instrument.devices.filters.FilterBank'>)
    325     return instance._signals[self.attr]
    326 except KeyError:
--> 327     return instance._instantiate_component(self.attr)
        Exception trying to inspect frame. No more locals available.

File /share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/ophyd/device.py:1361, in Device._instantiate_component(self=<class 'instrument.devices.filters.FilterBank'> instance, attr='transmission')
   1357             sig.subscribe(method, event_type=event_type, run=sig.connected)
   1358 except AttributeError as ex:
   1359     # Raise a different Exception, as AttributeError will be shadowed
   1360     # during initial access
-> 1361     raise RuntimeError(
        attr = 'transmission'
   1362         f"AttributeError while instantiating " f"component: {attr}"
   1363     ) from ex
   1365 return sig

RuntimeError: AttributeError while instantiating component: transmission

prjemian, 2024-11-16:
In 

> `transmission = TransmDerivedSignal(derived_from=attenuation, name="transmission", metadata=None)`

remove this part: `, metadata=None`

But then another error occurs.  Seems there is no example in ophyd to use a `DerivedSignal` as a `Component`.

@canismarko: Have you conquered this situation?

canismarko, 2024-11-16:
We used derived signals as components.

https://github.com/spc-group/haven/blob/7cddd5b09ef5ab6f4d618f4aa186c45a1ddaebc0/src/haven/devices/slits.py#L59

Or subclass the Derived Signal then use the derived Signal as a component.

https://github.com/spc-group/haven/blob/7cddd5b09ef5ab6f4d618f4aa186c45a1ddaebc0/src/haven/devices/xia_pfcu.py#L103

prjemian, 2024-11-16:
This ZIP file contains a notebook with a short demo:

[derived.zip](https://github.com/user-attachments/files/17787193/derived.zip)

prjemian, 2024-11-16:
Revised notebook on Mattermost:
https://mattermost.hzdr.de/bluesky/pl/4rbstmnjhbnrmcg8obwdgcyc9a

prjemian, 2024-11-16:
final form:

```py
    transmission = Component(TransmDerivedSignal, derived_from="attenuation")
```

jilavsky, 2024-11-16:
Excellent, missing "", kind of confusing, but it is name of the object, not object itself. Got it (for now). 
THANKS, works. I can. get transmission as expected.

closed 2024-11-16 by jilavsky