BCDA-APS/usaxs-bits #12: Error with custom Device using DerivedSignal

prjemian, 2024-11-23:
`AttributeError: 'FormattedComponent' object has no attribute 'metadata'`

<details>

```py
 File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 134, in plan_mutator
    msg = plan_stack[-1].send(ret)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 907, in fly_during_wrapper
    return (yield from plan2)
            ^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 181, in plan_mutator
    raise ex
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 134, in plan_mutator
    msg = plan_stack[-1].send(ret)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 181, in plan_mutator
    raise ex
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 134, in plan_mutator
    msg = plan_stack[-1].send(ret)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/bluesky/instrument/plans/axis_tuning.py", line 96, in tune_mr
    yield from _tune_base_(m_stage.r, md=md)
  File "/home/beams/USAXS/bluesky/instrument/plans/axis_tuning.py", line 72, in _tune_base_
    uuids = yield from axis.tune(md=md)     #note: the tune method comes from usaxs_motor_devices, TunableEpicsMotor, which uses lineup2
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/bluesky/instrument/devices/usaxs_motor_devices.py", line 109, in tune
    return (yield from _inner())
            ^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/bluesky/instrument/devices/usaxs_motor_devices.py", line 107, in _inner
    yield from self.post_tune_hook()
TypeError: 'NoneType' object is not iterable

class FilterBank(Device):
    """
    A single module of filters (12-blades).

    .. index:: Ophyd Device; FilterBank

    EXAMPLES::

        pfilter = FilterBank("ioc:", name="pfilter", bank="FL1")

    :see: 
    """

 
    class TransmDerivedSignal(DerivedSignal):
        def forward(self, value):
            return 1/value      
         
        #def inverse(self, value):
        #    return 1/value
        

    fPos = FormattedComponent(EpicsSignal, "{prefix}{_bank}:sortedIndex", kind="config")
    #control = FormattedComponent(EpicsSignalRO, "{prefix}bank{_bank}", string=True, kind="config")
    #bits = FormattedComponent(EpicsSignalRO, "{prefix}bitFlag{_bank}", kind="config")
    attenuation = FormattedComponent(EpicsSignalRO, "{prefix}{_bank}:attenuation_actual", kind="config")
    transmission = TransmDerivedSignal(derived_from=attenuation, name="transmission", metadata=None)

    def __init__(self, prefix, bank=None, **kwargs):
        self._bank = bank
        super().__init__(prefix, **kwargs)

ile /share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/ophyd/signal.py:630, in DerivedSignal.__init__(self=<class 'instrument.devices.filters.FilterBank.TransmDerivedSignal'> instance, derived_from=FormattedComponent(EpicsSignalRO, '{prefix}{_bank}:attenuation_actual', kind='config'), write_access=None, name='transmission', parent=None, **kwargs={'metadata': None})
    626     self._metadata_keys = getattr(derived_from, "metadata_keys", None)
    627     # And failing that, they are the defaults from all signals
    629 super().__init__(
--> 630     name=name, parent=parent, metadata=derived_from.metadata, **kwargs
        derived_from = FormattedComponent(EpicsSignalRO, '{prefix}{_bank}:attenuation_actual', kind='config')
        parent = None
        name = 'transmission'
        kwargs = {'metadata': None}
    631 )
    633 self._derived_from = derived_from
    635 self._allow_writes = write_access is not False

AttributeError: 'FormattedComponent' object has no attribute 'metadata'
```

</details>

prjemian, 2024-11-23:
Change this line:

```py
transmission = TransmDerivedSignal(derived_from=attenuation, name="transmission", metadata=None)
```

to

```py
   transmission = Component(TransmDerivedSignal, derived_from="attenuation")
```

prjemian, 2024-11-23:
Note: When `DerivedSignal` is used as a `Component`, the value of `derived_from=` is a text string (not a `Signal` object), with the name of the source component attribute.

prjemian, 2024-11-23:
@jilavsky Does this work for you?

jilavsky, 2024-11-25:
Works great, thanks.

closed 2024-11-25 by jilavsky