BCDA-APS/usaxs-bits #14: Remove logger messages

jilavsky, 2025-02-01:
After recent changes we have flurry of logger messages in the terminal.  These messages are now redirected to terminal. May be related to our change in lineup2? 
Currently I get both DEBUG:package: message and INFO:package:... 
I do not want to DEBUG messages in terminal and do not need the :package: in the message, if possible. This is not urgent, but annoying.

prjemian, 2025-02-14:
Sounds like a confusion (typo) in the use of  `logging.debug(...)` somewhere in the code.  Should be `logger.debug(...)`.  Same for `logging.info(...)`, should be `logger.info(...)`.

prjemian, 2025-02-14:
Searched, did not find any such lines.  This line is the likely cause: https://github.com/APS-USAXS/bluesky/blob/b6a583d37094b629c600044b11a9b8fda53bd2f1/instrument/usaxs_support/saveFlyData.py#L19

Suggest changing to a less-informative level, such as  `logging.WARNING`.

prjemian, 2025-02-14:
Not really a BUG, just a nuisance.

jilavsky, 2025-02-21:
I receive most logger.info messages twice. I check ed the code and here is example.. This line of code:

```python
logger.info(f"Finished SAXS/WAXS data collection in {elapsed} seconds.")
```

results in two messages to user. One would be more than enough...
```
I Fri-14:21:02 - Finished SAXS/WAXS data collection in 6.581214427947998 seconds.
INFO:instrument.plans.area_detector:Finished SAXS/WAXS data collection in 6.581214427947998 seconds.
```
How do we disable the second INFO message?

MDecarabas, 2025-05-05:
@jilavsky , I believe this is now solved. Can we close?