BCDA-APS/usaxs-bits #16: Communication with ar slow, data collection failed

jilavsky, 2025-02-10:
Sunday-to-Monday night (2/9-10), between 0:00 and 0:30am BS had issues with communication to a_stage.r when a_stage.r was not responding to BS. User attempted to run multiple tuning macros to fix the problem and reported that a_stage.r is not "responding properly". When I was called at 00:30am to investigate, a stage moved fine in epics. 

As result of these failures: 
AR became misaligned by 0.02deg, causing lineup2 in tune_ar to fail to find peak. 
Tunes - all of them (mr, a2rp, ar) - resulted in multiple BEC graphs spawned with one or more curves in each. Clearly somehow the code was not reusing old BEC graphs. 
Trying to close all those BEC graphs had generated graphical dialog with buttons Wait and Abort. It eventually hanged whole NoMachine GUI (X or Wyaland?) with no reaction to mouse clicks or keyboard commands. Looks like the desktop graphical interface (X or Wyaland?) has got in troubles. After minute or two graphical interface in NoMachine recovered and I was able to exit BS. That removed remaining BEC graphs. 
I have killed the terminal tab running BS and created a new one, restarted BS and all went well after that. 

Suspect network scanning?  Will send e-mail to IT.

jilavsky, 2025-02-14:
Tonight - around 1:17am - BlueSky stopped since it failed to move saxs_z stage and therefore hanged in move_WAXS code, resulting in loss of about 8 hours of beamtime. 

Actually, it hanged in move_SAXSOut(), line 120:

https://github.com/APS-USAXS/bluesky/blob/b6a583d37094b629c600044b11a9b8fda53bd2f1/instrument/plans/move_instrument.py#L120

When discovered in the morning, problem was still present. Recorded error messages, see below. 

I evaluated which line failed and was able to do .get() on terms.SAXS.z_out, so communication with the usxLAX PV holding the terms.SAXS.z_out was OK. What failed was moving the saxs_z stage.
NOTE: device was physically in SAXS mode, the saxs_z did NOT move at all from BS.

I was able to move the saxs_z stage from epics without any issues, no problems in communication using GUI. 

I tried to run other mode changing plans (mode_moveSAXSOut(), mode_OpenBeamPath()) and all failed with same error, but they all call same move_SAXSOut() plan.  

may be IMPORTANT observation:
exit from BS took obscenely long, well over minute. Something was taking forever to clean up. User and myself were staring at the missing command prompt and wondering what did we hanged. It took lot longer than I recall normally before, leading me to suspicion if this is resource leak or if we have issues closing record in DataBroker (re: DataBroker/Tiled [issues](https://github.com/APS-USAXS/bluesky/issues/19) ).  

Error messages:
```                                                                                                                                                                      
I Fri-01:17:46 - Finished SAXS/WAXS data collection in 2.5531346797943115 seconds.
INFO:instrument.plans.area_detector:Finished SAXS/WAXS data collection in 2.5531346797943115 seconds.
I Fri-01:17:46 - I0 value: 163558.0
INFO:instrument.plans.scans:I0 value: 163558.0
I Fri-01:17:46 - Found UsaxsSaxsMode = SAXS in beam
INFO:instrument.plans.mode_changes:Found UsaxsSaxsMode = SAXS in beam
I Fri-01:17:46 - Moving to proper WAXS mode
                                                                                                                                                                      
I Fri-01:17:46 - Moving SAXS out of beam
INFO:instrument.plans.move_instrument:Moving SAXS out of beam
WARNING:ophyd:Status(obj=EpicsSignal(read_pv='usxLAX:SAXS:USAXSSAXSMode', name='terms_SAXS_UsaxsSaxsMode', parent='terms_SAXS', value=65535, timestamp=1739517466.44927, auto_monitor=True, string=False, write_pv='usxLAX:SAXS:USAXSSAXSMode', limits=False, put_complete=True), done=False, success=False) has timed out
ERROR:bluesky:Run aborted
ophyd.utils.errors.StatusTimeoutError: Status Status(obj=EpicsSignal(read_pv='usxLAX:SAXS:USAXSSAXSMode', name='terms_SAXS_UsaxsSaxsMode', parent='terms_SAXS', value=65535, timestamp=1739517466.44927, auto_monitor=True, string=False, write_pv='usxLAX:SAXS:USAXSSAXSMode', limits=False, put_complete=True), done=False, success=False) failed to complete in specified timeout.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/run_engine.py", line 1580, in _run
    msg = self._plan_stack[-1].throw(stashed_exception or resp)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 1370, in __call__
    return (yield from plan)
            ^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 1223, in baseline_wrapper
    return (yield from plan_mutator(plan, insert_baseline))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 89, in plan_mutator
    msg = plan_stack[-1].throw(exception)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 845, in monitor_during_wrapper
    return (yield from plan2)
            ^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 89, in plan_mutator
    msg = plan_stack[-1].throw(exception)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 89, in plan_mutator
    msg = plan_stack[-1].throw(exception)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 907, in fly_during_wrapper
    return (yield from plan2)
            ^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 89, in plan_mutator
    msg = plan_stack[-1].throw(exception)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/share1/bluesky/conda_envs/bluesky_2024_3/lib/python3.11/site-packages/bluesky/preprocessors.py", line 89, in plan_mutator
    msg = plan_stack[-1].throw(exception)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/bluesky/user/linkam.py", line 212, in fanLinkamPlan
    yield from collectAllThree(isDebugMode)
[... cut ...]
    raise FailedStatus(ret) from exc
bluesky.utils.FailedStatus: Status(obj=EpicsSignal(read_pv='usxLAX:SAXS:USAXSSAXSMode', name='terms_SAXS_UsaxsSaxsMode', parent='terms_SAXS', value=65535, timestamp=1FailedStatus: Status(obj=EpicsSignal(read_pv='usxLAX:SAXS:USAXSSAXSMode', name='terms_SAXS_UsaxsSaxsMode', parent='terms_SAXS', value=65535, timestamp=1739517466.44927, auto_monitor=True, string=False, write_pv='usxLAX:SAXS:USAXSSAXSMode', limits=False, put_complete=True), done=True, success=False)

In [39]: 
```

prjemian, 2025-02-14:
> exit from BS took obscenely long, well over minute. Something was taking forever to clean up.

Has been observed.  Some code has many objects left for removal.  Removal is forced by [`atexit()`](https://docs.python.org/3/library/atexit.html) handling.

jilavsky, 2025-05-28:
Restarting BS code every ~24 hours was sufficient. Needs to be tested on BITS instrument. Closing.

closed 2025-05-28 by jilavsky