BCDA-APS/usaxs-bits #18: Databroker issues

jilavsky, 2025-02-12:
Looking in tiled, we need to fix stuff - this is for Flyscan data:
Databroker contains incorrect information. Instrument name is cool, but let's call this "APS USAXS instrument".  This was not collected under commissioning and APSBSS had proper ESAF and GUP loaded, we need to use value from APSBSS. 

I need to know where this defined to fix it. Cursory check of bluesky code does not find Most Glorious Scientific Instrument and my templates also do not have it. Where is it? 

            "RUN_ENGINE": {
               "DEFAULT_METADATA": {
                  "beamline_id": "instrument",
                  "databroker_catalog": "temp",
                  "instrument_name": "Most Glorious Scientific Instrument",
                  "proposal_id": "commissioning"
               },
               "USE_PROGRESS_BAR": false
            },
            "SPEC_DATA_FILES": {
               "FILE_EXTENSION": "dat"
            },
            "XMODE_DEBUG_LEVEL": "Minimal"
         },
         "instrument_name": "Most Glorious Scientific Instrument",
         "login_id": "usaxs@usaxscontrol.xray.aps.anl.gov",
         "pid": 1099109,
         "plan_args": {
            "pos_X": 0,
            "pos_Y": 0,
            "scan_title": "TPEFeedstock80_21C_121min",
            "thickness": 1.155
         },

prjemian, 2025-02-12:
You should be able to set the `proposal_id` with the latest value from `apsbss` by direct assignment:

`RE.md["proposal_id"] = <proposal ID value from apsbss>`

jilavsky, 2025-02-12:
my iconfig.yml has different values. I also found some values in ~/Bluesky_RunEngine_md/ folder. But it also doe snot have the "Most Glorious Scientific Instrument". I cannot find it in my Github or anywhere else, so where does that come from? 
I can overwrite this in newUser which is compulsory to run (as Fan found, nothing works without it).

jilavsky, 2025-02-12:
The iconfig.yml has wrong DATABROKER_CATALOG (20idb_usaxs) and we are suing the correct one, so that file is obsolete and somethign else overwrites it alter.

prjemian, 2025-02-14:
In addition to the `iconfig.yml` settings, which are used when restoring `RE.md` is not successful, all settings are stored by `PersistentDict` in directory `~/.config/Bluesky_RunEngine_md/`.  You can change the values that will be stored by changing the values in `RE.md`.  Such as:

```py
RE.md["instrument_name"] = "APS 12-ID-E USAXS"
```

Some advice about some these settings is [provided](https://github.com/BCDA-APS/BITS/tree/a48d6ccd9ebe373e5835b7128413de88c6545055?tab=readme-ov-file#how-to-use-the-template) in newer instrument setups.  Here are the settings relevant for USAXS:

- DATABROKER_CATALOG: change from temp to your catalog's name
- beamline_id: **one word** beamline name (such as known by **APS scheduling system**)
- instrument_name: descriptive name of your beamline

jilavsky, 2025-02-15:
This final suggestion worked for me great. Closing.

closed 2025-02-15 by jilavsky