BCDA-APS/usaxs-bits #19: Tiled access to Databroke is painfully slow again

jilavsky, 2025-02-12:
Databroker access through tiled - using Gemviz or Igor - seems to be again impossible slow. Getting list of 100 scans from 2/9 should take few seconds, takes 30 seconds (using Igor or Gemviz).  We had this in January and created new catalog. This is not really usable for humans. Something is wrong, again, with the Databroker or Tiled server. 
I tried on 20idb_usaxs_retired and it takes about 5 seconds in Gemviz. That is usable.

jilavsky, 2025-02-12:
Note, it is slow from web interface also, as expected...

prjemian, 2025-02-14:
## First question
_Where, exactly, is the problem?_

### databroker?

Develop a test program and run it on various workstations (where credentials are available).

Test program:

```py
"""Explore the "usaxs" catalog.  Why is it slow again?"""

import time

def main():
    t0 = time.monotonic()
    dt = time.monotonic() - t0
    print(f"No-op: {dt=}s")

    t0 = time.monotonic()
    import databroker  # noqa
    dt = time.monotonic() - t0
    print(f"import databroker: {dt=}s")

    t0 = time.monotonic()
    catalogs = list(databroker.catalog)
    dt = time.monotonic() - t0
    print(f"{catalogs=}: {dt=}s")

    t0 = time.monotonic()
    cat = databroker.catalog["usaxs"]
    dt = time.monotonic() - t0
    print(f"{dt=}s")
    print(cat)

    t0 = time.monotonic()
    num_runs = len(cat)
    dt = time.monotonic() - t0
    print(f"{num_runs=}: {dt=}s")

if __name__ == "__main__":
    main()
```

prjemian, 2025-02-14:
#### results from workstation: `wow`

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Pete/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Pete/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Reported times from test program |   |   |  
-- | -- | -- | --
  |   |   |   |   |  
run | no-op | import | cat | #runs | time
1 | 4.78E-07 | 46.14459 | 12.66857 | 14787 | 0.110852
2 | 2.29E-07 | 1.15E-06 | 0.224809 | 14789 | 0.028435
3 | 1.77E-07 | 1.40E-06 | 0.202008 | 14789 | 0.019148

</body>

</html>

(as copy/paste from Excel)

prjemian, 2025-02-14:
#### Results from workstation `wow`

Reported times (s) from test program:

run | no-op | import | cat | #runs | time
--- | --- | --- | --- | --- | ---
1 | 4.78E-07 | 46.14459196 | 12.66856876 | 14787 | 0.110851771
2 | 2.29E-07 | 1.15E-06 | 0.2248085 | 14789 | 0.028435119
3 | 1.77E-07 | 1.40E-06 | 0.202008152 | 14789 | 0.019148354

prjemian, 2025-02-14:
Python test code for searching runs via tiled server (using some search functions vendored from gemviz, but not shown here):

```py
def timings_tiled(catalog="usaxs", uri="http://localhost:8020"):
    t0 = time.monotonic()
    t_noop = time.monotonic() - t0
    print(f"No-op: {t_noop=}s")

    t0 = time.monotonic()
    import tiled.client, tiled.queries  # noqa

    t_import = time.monotonic() - t0
    print(f"import tiled client and queries: {t_import=}s")

    t0 = time.monotonic()
    client = tiled.client.from_uri(uri, "dask")
    t_client = time.monotonic() - t0
    print(f"{uri=} {t_client=}s")

    cat = client[catalog]
    t_cat = time.monotonic() - t0
    print(f"{uri=} {catalog=} {cat=} {t_cat=}s")

    t0 = time.monotonic()
    num_runs = len(cat)
    t_num = time.monotonic() - t0
    print(f"{num_runs=}: {t_num=}s")

    # TODO: search for specific scans

    # if since is not None:
    #     cat = cat.search(QueryTimeSince(since))
    # if until is not None:
    #     cat = cat.search(QueryTimeUntil(until))

    # for k, v in keys.items():
    #     cat = cat.search(tiled.queries.Key(k) == v)

    # for v in text:
    #     cat = cat.search(tiled.queries.FullText(v, case_sensitive=False))
    # for v in text_case:
    #     cat = cat.search(tiled.queries.FullText(v, case_sensitive=True))

    t0 = time.monotonic()
    subcat = cat.search(QueryTimeSince("2025-02-05"))
    subcat = subcat.search(QueryTimeUntil("2025-02-12"))
    subcat = subcat.search(tiled.queries.Key("plan_name") == "tune_a2rp")
    t_search = time.monotonic() - t0
    print(f"{subcat=}: {t_search=}s")

    print(
        f"{uri} | {t_noop:.2e} | {t_import:.2e} | {t_cat:.2e} | {t_client:.2e} | {num_runs} | {t_num:.2e} | {t_search:.2e}"
    )
```

prjemian, 2025-02-14:
### Search for runs via tiled server

Search details:

```py
    subcat = cat.search(QueryTimeSince("2025-02-05"))
    subcat = subcat.search(QueryTimeUntil("2025-02-12"))
    subcat = subcat.search(tiled.queries.Key("plan_name") == "tune_a2rp")
```

#### Results from workstation `otz`

Reported times (s) from test program using tiled server on `http://localhost:8020`:

run | no-op | import | client | cat | #runs | time | search
--- | --- | --- | --- | --- | --- | --- | ---
1 | 3.01e-07 | 7.80e-01 | 2.34e-01 | 2.16e-01 | 14999 | 3.32e-06 | 2.48e-04
2 | 2.76e-07 | 7.44e-01 | 1.85e-01 | 1.66e-01 | 15005 | 3.95e-06 | 2.57e-04
3 | 3.91e-07 | 3.17e-01 | 1.14e-01 | 9.67e-02 | 15005 | 2.46e-06 | 1.73e-04

#### Results from workstation `usaxscontrol`

Reported times (s) from test program using tiled server on `http://localhost:8020`:

run | no-op | import | client | cat | #runs | time | search
--- | --- | --- | --- | --- | --- | --- | ---
1 | 3.47e-07 | 1.51e+00 | 3.74e-01 | 3.60e-01 | 15006 | 1.33e-06 | 1.28e-04
2 | 3.71e-07 | 1.42e-01 | 5.27e-02 | 3.91e-02 | 15008 | 1.76e-06 | 1.52e-04
3 | 2.46e-07 | 1.36e-01 | 5.70e-02 | 4.42e-02 | 15010 | 1.34e-06 | 1.24e-04

### Conclusions

- Initial imports are slower, same as before
- The original problem is not due to `tiled` (as tested here)

prjemian, 2025-02-14:
@jilavsky : Can you upload the Igor code you use to access scans?  I suspect there are differences from the tests shown above.

jilavsky, 2025-02-14:
Igor code 
```
Function testTiled()
	//this tests two different catalogs available on usaxscontrol 
	//it asks for listing of records. By default, we should get 100 data sets in json.
	//these addresses can be opend in web browser with same effect.
	//code below ignores response, dumps obtained data,
        //         and only measures how long it takes to get the server response. 
	
	//test 1: old database created in 2023 when instrument was at 120IDB
	//note: this database works fine
	
	print "Testing Databroker/Tiled dbs"
	string TempAddress = "http://usaxscontrol.xray.aps.anl.gov:8020/api/v1/search/20idb_usaxs_retired_2023-12-05"
	Variable timerRefNum, microSeconds
	timerRefNum = StartMSTimer
	URLRequest/Z url=TempAddress
	if(V_Flag!=0)
		abort "Server not available"
	endif
	microSeconds = StopMSTimer(timerRefNum)
	Print microSeconds/1e6, "secs to get first 100 records from "+TempAddress
	print "***"
	
	//test 2: new database created in January 2025 and used currently. 
	TempAddress = "http://usaxscontrol.xray.aps.anl.gov:8020/api/v1/search/usaxs"
	timerRefNum = StartMSTimer
	URLRequest/Z url=TempAddress
	if(V_Flag!=0)
		abort "Server not available"
	endif
	microSeconds = StopMSTimer(timerRefNum)
	Print microSeconds/1e6, "secs to get first 100 records from "+TempAddress
	print "***"
end
```

result:
```
testTiled()
  Testing Databroker/Tiled dbs
  0.518847  secs to get first 100 records from http://usaxscontrol.xray.aps.anl.gov:8020/api/v1/search/20idb_usaxs_retired_2023-12-05
  ***
  89.8739  secs to get first 100 records from http://usaxscontrol.xray.aps.anl.gov:8020/api/v1/search/usaxs
  ***
```

prjemian, 2025-02-14:
### Search for runs with Python `requests` via tiled server

This client prepares a URL (server URI, catalog, and search parameters) and requests JSON results from the tiled server.

```py
def timings_requests_tiled(catalog="usaxs", uri="http://localhost:8020"):
    import requests

    t0 = time.monotonic()
    t_noop = time.monotonic() - t0
    # print(f"No-op: {t_noop=}s")

    url = f"{uri}/api/v1/search/{catalog}"
    t0 = time.monotonic()
    r = requests.get(url)
    t_request = time.monotonic() - t0

    print(f"{url} | {t_noop:.2e} | {t_request:.2e}")
```

#### Results from testing

These are the times reported from the above test code:

search URL | no-op | search
--- | --- | ---
http://otz:8020/api/v1/search/developer | 3.73e-07 | 9.44e-02
http://otz:8020/api/v1/search/usaxs | 2.25e-07 | 9.26e+01
http://otz:8020/api/v1/search/developer | 2.82e-07 | 3.20e-01
http://otz:8020/api/v1/search/20idb_usaxs_retired_2023-12-05 | 1.96e-07 | 7.75e-03
http://otz:8020/api/v1/search/usaxs | 1.65e-07 | 9.52e+01
http://otz:8020/api/v1/search/developer | 4.45e-07 | 2.07e-01
http://otz:8020/api/v1/search/usaxs | 2.01e-07 | 9.61e+01
http://otz:8020/api/v1/search/20idb_usaxs_retired_2023-12-05 | 1.83e-07 | 7.61e-03
http://otz:8020/api/v1/search/usaxs_12idc_retired_2025-01-09 | 1.45e-07 | 8.14e+01
http://otz:8020/api/v1/search/usaxs_9idc_retired_2022-08-10 | 2.80e-07 | 5.47e-01

### Conclusion

- This test reproduces reported problem catalogs (`usaxs` and `usaxs_12idc_retired_2025-01-09`).
- Previous tests did not ask for list of runs, just quantity.

prjemian, 2025-02-14:
The time it takes to retrieve basic information from a catalog is of no significant impediment.  Once a catalog is opened, it takes significant time to retrieve information about a run (any run) from certain catalogs (`usaxs` and `usaxs_12idc_retired_2025-01-09`).

Modifying the above databroker test, by adding code that gets the last run, took ~6s to retrieve the last run from the problem catalogs while <0.1s for the others.

prjemian, 2025-02-14:
Learning how to increase the default 30s timeout for the `tiled.client` ... as needed for the problem catalogs.

prjemian, 2025-02-14:
The slow read performance is not affecting the ability of the instrument to write new data.  The instrument has added >400 new runs since 6 hours ago.

prjemian, 2025-02-14:
@danielballan  We're looking at the second time a (MongoDB) collection for USAXS has started to exhibit _painfully slow_ read response.  Any ideas?

danielballan, 2025-02-15:
Just caught up. Let’s hop on a Zoom next week and sort this out. Nothing substantively changed, so my guess is this is some regression that we can easily revert with a little attention.

prjemian, 2025-02-21:
Looked at problem catalog `usaxs_12idc_retired_2025-01-09` in MongoDB.

For some reason, using `mongosh`, we saw it was missing its indices:

```mongosh
usaxs-bluesky> db.run_start.getIndexes()
[
  {
    v: 2,
    key: { _id: 1 },
    name: '_id_',
    ns: 'usaxs-bluesky.run_start'
  }
]
```

After the repair procedure, there were many indices, such as `scan_id`, `time`, and `uid`.

The repair procedure used `bluesky_tiled_plugins`, `ipython`, `suitcase_mongo-0.7.0`, and `tiled-0.1.0b17`:

```bash
conda create -y -n test tiled
conda activate test
pip install --pre ipython suitcase-mongo bluesky_tiled_plugins
```

```python
import suitcase.mongo_normalized
from tiled.client import from_uri

cat = from_uri("http://otz:8020", timeout=180)["usaxs_12idc_retired_2025-01-09"]
serializer = suitcase.mongo_normalized.Serializer(metadata_uri, asset_uri)
serializer.create_indexes()
```

prjemian, 2025-02-21:
@danielballan :heart:  Big thanks for solving this!

prjemian, 2025-02-21:
I imagine something we're doing got rid of the indices.  Don't know what.  Become sensitive to the response time.  If it becomes slow again, try to recall what happened just before.  Me, too.

prjemian, 2025-02-24:
Repaired.