BCDA-APS/usaxs-bits #26: Make the bits automatically install devices and palns

jilavsky, 2025-05-03:
Start requires now importing devices and plans manually. How do I add this to be automatic? Annoying beyond reason. Too complicated commands.

prjemian, 2025-05-03:
`thing = oregistry["thing"]` for each thing you need in a plan

jilavsky, 2025-05-03:
Not what I meant. Staring BITS: 
ipython, wait. 
from ... import * #imports all devices, wait
from ... import plans #import plans. 
Where do I put the two imports so they get done automatically? 
in startup.py?

jilavsky, 2025-05-04:
OK, adding "from .plans import *" to load all plans to startup.py loads plans. How do I trigger "from usaxs.startup import *" when I start ipython? Old BlueSky did load stuff automatically.

MDecarabas, 2025-05-05:
Load startup.py as default when opening ipython

prjemian, 2025-05-05:
`ipython -i -c from usaxs.startup import *` is the startup command equivalent to what you have done in the past.

prjemian, 2025-05-05:
Do you want this wrapped in a shell script (with conda environment activation included)?

jilavsky, 2025-05-05:
Shell script would be great. It gets really annoying quickly when I had to restart the BITS every 2-3 minutes.

prjemian, 2025-05-07:
@MDecarabas : Planning to create a bash starter script (`start-usaxs-bits.sh`).  What directory should that script be added here?

ravescovi, 2025-05-07:
On the top level `/scripts` would be my choice

MDecarabas, 2025-05-07:
Agree with @ravescovi

prjemian, 2025-05-07:
Ok.  Place it in a **new** `scripts/` directory at the root level, here?

[image]

prjemian, 2025-05-07:
We should also add any and all `.DS_Store` files to `.gitignore`.

ravescovi, 2025-05-07:
Yes. I think the *new* `$TOP/scripts/` is the correct way.
Agreed on .DS_Store, not sure why it keeps going through.

prjemian, 2025-05-07:
> not sure why it keeps going through.

I'll make a branch with the new directory and shell script and the change(s) to the `.gitignore` file.

MDecarabas, 2025-05-07:
> > not sure why it keeps going through.
> 
> I'll add to the `bits` branch with the new directory and shell script and the change(s) to the `.gitignore` file.

pretty sure, that the .gitignore file come from bits-starter. Should be changed there

prjemian, 2025-05-07:
When the branch is cloned locally, the `.DS_Store` file does not appear.  Confirmed the file is already in `.gitignore`.  Asked AI how to remove it.  Summary:

`git rm --cached '*.DS_Store'`

prjemian, 2025-05-07:
`.DS_Store` file(s) removed in commit e8a73647bed2

MDecarabas, 2025-05-07:
sure, let's make sure this doesn't happen in the future by fixing the root problem as well.

prjemian, 2025-05-07:
Would an organization-wide `.gitignore` file with this file name (and other well-known entries) fix the root problem?

ravescovi, 2025-05-07:
This is a good question. I have no idea, probably interesting to start pushing the org wide behaviours.

prjemian, 2025-05-07:
Got to be careful with the BCDA-APS org since there are many projects that do not use Python.

ravescovi, 2025-05-07:
There is no damage on a very large and complete .gitignore =)

prjemian, 2025-05-07:
@jilavsky Just added the starter `scripts/start-usaxs-bits.sh` which should start the instrument session same as you have been.  You could copy or link it in `~/bin` so it would be callable from any working directory.  Edit these lines as needed:
https://github.com/BCDA-APS/usaxs-bits/blob/65a9d2dc0979e2285c9844b88952adc61c21ff74/scripts/start-usaxs-bits.sh#L6-L7

jilavsky, 2025-05-07:
I just checked... 
Conda environment for bits is called "bits_usaxs". There is no bluesky_2025_1.

jilavsky, 2025-05-07:
Done. Environment set. Closing. Thanks!

closed 2025-05-07 by jilavsky