BCDA-APS/usaxs-bits #30: lineup2 in tune_ar fails when called with scaler0

jilavsky, 2025-05-23:
The following call (which used to work on old BS USAXS) generates RE error:
```python 
        scaler0.select_channels(["PD_USAXS"])
        stats = SignalStatsCallback()
        yield from lineup2(
            [scaler0],
            a_stage.r,
            -a_stage.r.tune_range.get(),
            a_stage.r.tune_range.get(),
            31,
            nscans=1,
            signal_stats=stats,
            md=md,
        )
```
Errors below. Best guess, label_stream_wrapper cannot deal with scaler0, looks like something is missing. Do we need to fix scaler0 definition somhow? 

Errors :
In [1]: RE(tune_ar())
I Fri-11:06:00.943: tuning axis: a_stage_r

Transient Scan ID: 3     Time: 2025-05-23 11:06:01
Persistent Unique Scan ID: '693a11d6-1542-4d81-a671-1de8637e939b'
New stream: 'label_start_motor'
New stream: 'primary'
+-----------+------------+------------+------------+
|   seq_num |       time |  a_stage_r |   PD_USAXS |
+-----------+------------+------------+------------+
|         1 | 11:06:03.0 |  1.2372391 |      12763 |
|         2 | 11:06:03.2 |  1.2372921 |      12998 |
|         3 | 11:06:03.4 |  1.2373458 |        164 |
|         4 | 11:06:03.6 |  1.2373985 |      16395 |
|         5 | 11:06:03.8 |  1.2374513 |      14139 |
|         6 | 11:06:03.9 |  1.2375057 |      13115 |
|         7 | 11:06:04.1 |  1.2375588 |      12778 |
|         8 | 11:06:04.3 |  1.2376119 |      12820 |
|         9 | 11:06:04.5 |  1.2376671 |        146 |
|        10 | 11:06:04.7 |  1.2377182 |        222 |
|        11 | 11:06:04.9 |  1.2377719 |        202 |
|        12 | 11:06:05.1 |  1.2378246 |      16160 |
|        13 | 11:06:05.2 |  1.2378793 |        159 |
|        14 | 11:06:05.4 |  1.2379327 |      15940 |
|        15 | 11:06:05.6 |  1.2379849 |      13999 |
|        16 | 11:06:05.8 |  1.2380382 |      13136 |
|        17 | 11:06:06.0 |  1.2380932 |      12700 |
|        18 | 11:06:06.2 |  1.2381449 |      12578 |
|        19 | 11:06:06.3 |  1.2381993 |      12379 |
|        20 | 11:06:06.5 |  1.2382518 |      12271 |
|        21 | 11:06:06.7 |  1.2383074 |      11992 |
|        22 | 11:06:06.9 |  1.2383579 |      12765 |
|        23 | 11:06:07.1 |  1.2384121 |      13181 |
|        24 | 11:06:07.3 |  1.2384654 |      12792 |
|        25 | 11:06:07.4 |  1.2385179 |      12932 |
|        26 | 11:06:07.6 |  1.2385718 |      12892 |
|        27 | 11:06:07.8 |  1.2386249 |      13158 |
|        28 | 11:06:08.0 |  1.2386779 |      13125 |
|        29 | 11:06:08.2 |  1.2387324 |       1318 |
|        30 | 11:06:08.4 |  1.2387852 |      16416 |
|        31 | 11:06:08.6 |  1.2388385 |        147 |
+-----------+------------+------------+------------+
generator tune_ar ['693a11d6'] (scan num: 3)

E Fri-11:06:08.790: Error in tune_ar: 'scaler0'
E Fri-11:06:08.790: Run aborted
Traceback (most recent call last):
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py", line 1609, in _run
    msg = self._plan_stack[-1].send(resp)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/plans/labels_to_streams.py", line 231, in label_stream_wrapper
    return (yield from bpp.plan_mutator(plan, action[when.value]))
[... 438 lines cut ...]
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py:206 in plan_mutator
    inner_ret = yield msg
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py:206 in plan_mutator
    inner_ret = yield msg
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1676 in _run
    new_response = await coro(msg)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1900 in _close_run
    ret = await current_run.close_run(msg)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/bundlers.py:165 in close_run
    await self.emit(DocumentNames.stop, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:2679 in emit
    self.emit_sync(name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:2676 in emit_sync
    self.dispatcher.process(name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:2699 in process
    exceptions = self.cb_registry.process(name, name.name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:422 in process
    func(*args, **kwargs)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:512 in __call__
    return mtd(*args, **kwargs)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/scan_signal_statistics.py:157 in receiver
    getattr(self, key)(document)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/scan_signal_statistics.py:201 in stop
    self._data[self._detectors[0]],
KeyError: 'scaler0'

prjemian, 2025-05-23:
In the list of detectors, name the specific scaler signal first, then the entire scaler device.

```py
        yield from lineup2(
            [UPD_SIGNAL, scaler0],
            a_stage.r,
            -a_stage.r.tune_range.get(),
            a_stage.r.tune_range.get(),
            31,
            nscans=1,
            signal_stats=stats,
            md=md,
        )
```

Without this, `lineup2` will use the first channel it finds.  Likely to be the constant count time.  Very flat.

prjemian, 2025-05-23:
BTW, I'm starting on the amplifiers this AM.

- #27

This work will touch these files in some way or another, probably more.  Draft PR soon, certainly not ready this morning.

prjemian, 2025-05-23:
[image]

prjemian, 2025-05-23:
[image]

prjemian, 2025-05-23:
First "detector" is the signal to use for lineup.  The "scaler0" is needed _after the first detector_ so that the scaler is triggered properly at each step.

jilavsky, 2025-05-23:
That has broken the lineup2 even more. I'll wait until you are done, I am touching same files. Too much mess. We will fix it after amplifier is fixed.

closed 2025-05-28 by jilavsky