BCDA-APS/usaxs-bits #41: Cannot make lineup2 detectors working.

jilavsky, 2025-05-25:
tune_ar. 
lineup2 needs list of detectors. 

Last PRJ suggestion I recall was that it should be:
[UPD_SIGNAL, scaler0]  --- that fails miserably during counting of first point. See messages below. 
Other variations:
[UPD, scaler0] --- fails with same errors (very large number of lines looks very similar) 

only [scaler0] - works best, finishes the scan and then throws error: 
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/scan_signal_statistics.py:201 in stop
    self._data[self._detectors[0]],

***

For [UPD_SIGNAL, scaler0] these are the errors:

```
In [1]: RE(tune_ar())
I Sun-13:51:40.265: tuning axis: a_stage_r

Transient Scan ID: 20     Time: 2025-05-25 13:51:41
Persistent Unique Scan ID: 'b31e8698-f842-4397-b2ef-acd4cb74fdd4'
New stream: 'label_start_motor'
New stream: 'primary'
+-----------+------------+------------+------------+
|   seq_num |       time |  a_stage_r |   PD_USAXS |
+-----------+------------+------------+------------+
+-----------+------------+------------+------------+
generator tune_ar ['b31e8698'] (scan num: 20)

E Sun-13:51:42.001: Error in tune_ar: None
E Sun-13:51:42.002: Run aborted
Traceback (most recent call last):
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 604, in contingency_wrapper
    ret = yield from plan
          ^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/plans.py", line 1166, in inner_scan_nd
    yield from per_step(detectors, step, pos_cache)
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py", line 1952, in __iter__
    return (yield from self._iter)
            ^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/plan_stubs.py", line 1729, in one_nd_step
    yield from take_reading(list(detectors) + list(motors))  # type: ignore  # Movable issue
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py", line 1952, in __iter__
    return (yield from self._iter)
            ^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/plan_stubs.py", line 1471, in trigger_and_read
    return (yield from rewindable_wrapper(inner_trigger_and_read(), rewindable))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 731, in rewindable_wrapper
    return (yield from plan)
            ^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/plan_stubs.py", line 1466, in inner_trigger_and_read
    ret = yield from contingency_wrapper(read_plan(), except_plan=exception_path, else_plan=standard_path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 623, in contingency_wrapper
    yield from else_plan()
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/plan_stubs.py", line 1460, in standard_path
    yield from save()
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py", line 1952, in __iter__
    return (yield from self._iter)
            ^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/plan_stubs.py", line 128, in save
    return (yield Msg("save"))
            ^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 206, in plan_mutator
    inner_ret = yield msg
                ^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 259, in msg_mutator
[... 566 lines cut ...]
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1676 in _run
    new_response = await coro(msg)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1900 in _close_run
    ret = await current_run.close_run(msg)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/bundlers.py:165 in close_run
    await self.emit(DocumentNames.stop, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:2679 in emit
    self.emit_sync(name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:2676 in emit_sync
    self.dispatcher.process(name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:2699 in process
    exceptions = self.cb_registry.process(name, name.name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:422 in process
    func(*args, **kwargs)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:512 in __call__
    return mtd(*args, **kwargs)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/scan_signal_statistics.py:157 in receiver
    getattr(self, key)(document)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/scan_signal_statistics.py:200 in stop
    self._data[self._x_name],
KeyError: None

In [2]: 

```

jilavsky, 2025-05-25:
Is any one of our detectors actually detector as BS expects? 
UPD is EpicsSignalRO
UPD_SIGNAL is ScalerChannel
I would assume that UPD_SIGNAL is our UPD detector and can be counted, but lineup2 clearly does not think so...

prjemian, 2025-05-25:
The objects without `_SIGNAL are the right ones. Needs investigation from command line...

jilavsky, 2025-05-25:
OK, this works:

RE(lineup2([UPD],s_stage.x, -0.1, 0.1, 31,nscans=1))

let me try without scaler0 as this which worked...

jilavsky, 2025-05-25:
Yep, that is solution. 
No scaler0 in the command and use only UPD in deetctors. I can work with that. 
prior BS code used scaler.select and scaler0...

jilavsky, 2025-05-25:
Looking at the errors:
```  
...descriptor_doc, compose_event, d_objs = await self._prepare_stream(desc_key, objs_dks)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/bundlers.py:207 in _prepare_stream
    await self.emit(DocumentNames.descriptor, self._descriptors[desc_key].descriptor_doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:2679 in emit
    self.emit_sync(name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:2676 in emit_sync
    self.dispatcher.process(name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:2699 in process
    exceptions = self.cb_registry.process(name, name.name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:422 in process
    func(*args, **kwargs)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:512 in __call__
    return mtd(*args, **kwargs)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/callback_base.py:94 in receiver
    handler(doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/spec_file_writer.py:794 in descriptor
    self.data_labels = get_data_labels()
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/spec_file_writer.py:786 in get_data_labels
    dets, others = parse(self.detectors)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/spec_file_writer.py:772 in parse
    hints = doc["hints"][k_obj]["fields"]
KeyError: 'PD_USAXS'
```

it looks like this is failing in spec_flie_writer? 
Also, PD_USAXS is nickname used only in epics and then somehow called here.

jilavsky, 2025-05-25:
Maybe we need to add to scaler0.chan04 something so code does not fail. It used to work, so our scaler must be somehow different than before.

jilavsky, 2025-05-25:
I started old BS to check what ths scaler looks like and they are exactly the same! see below
```
oldBS:
In [1]: scaler0
Out[1]: ScalerCH(prefix='usxLAX:vsc:c0', name='scaler0', read_attrs=['channels', 'channels.chan01', 
'channels.chan01.s', 'channels.chan02', 'channels.chan02.s', 'channels.chan03', 'channels.chan03.s', 
'channels.chan04', 'channels.chan04.s', 'channels.chan05', 'channels.chan05.s', 'time'], 
configuration_attrs=['channels', 'channels.chan01', 'channels.chan01.chname', 'channels.chan01.preset', 
'channels.chan01.gate', 'channels.chan02', 'channels.chan02.chname', 'channels.chan02.preset', 
'channels.chan02.gate', 'channels.chan03', 'channels.chan03.chname', 'channels.chan03.preset', 
'channels.chan03.gate', 'channels.chan04', 'channels.chan04.chname', 'channels.chan04.preset', 
'channels.chan04.gate', 'channels.chan05', 'channels.chan05.chname', 'channels.chan05.preset', 
'channels.chan05.gate', 'count_mode', 'delay', 'auto_count_delay', 'freq', 'preset_time', 'auto_count_time', 
'egu'])

In [2]: scaler0.select_channels(["PD_USAXS"])
In [3]: scaler0
Out[3]: ScalerCH(prefix='usxLAX:vsc:c0', name='scaler0', read_attrs=['channels', 
'channels.chan04', 'channels.chan04.s', 'time'], configuration_attrs=['channels', 
'channels.chan04', 'channels.chan04.chname', 'channels.chan04.preset', 'channels.chan04.gate', 
'count_mode', 'delay', 'auto_count_delay', 'freq', 'preset_time', 'auto_count_time', 'egu'])

new BS:

Out[7]: ScalerCH(prefix='usxLAX:vsc:c0', name='scaler0', read_attrs=['channels', 'channels.chan01', 
'channels.chan01.s', 'channels.chan02', 'channels.chan02.s', 'channels.chan03', 'channels.chan03.s', 
'channels.chan04', 'channels.chan04.s', 'channels.chan05', 'channels.chan05.s', 'time'], 
configuration_attrs=['channels', 'channels.chan01', 'channels.chan01.chname', 'channels.chan01.preset', 
'channels.chan01.gate', 'channels.chan02', 'channels.chan02.chname', 'channels.chan02.preset', 
'channels.chan02.gate', 'channels.chan03', 'channels.chan03.chname', 'channels.chan03.preset', 
'channels.chan03.gate', 'channels.chan04', 'channels.chan04.chname', 'channels.chan04.preset', 
'channels.chan04.gate', 'channels.chan05', 'channels.chan05.chname', 'channels.chan05.preset', 
'channels.chan05.gate', 'count_mode', 'delay', 'auto_count_delay', 'freq', 'preset_time', 'auto_count_time', 
'egu'])

In [8]: scaler0.select_channels(["PD_USAXS"])
In [9]: scaler0
Out[9]: ScalerCH(prefix='usxLAX:vsc:c0', name='scaler0', read_attrs=['channels', 
'channels.chan04', 'channels.chan04.s', 'time'], configuration_attrs=['channels', 
'channels.chan04', 'channels.chan04.chname', 'channels.chan04.preset', 'channels.chan04.gate', 
'count_mode', 'delay', 'auto_count_delay', 'freq', 'preset_time', 'auto_count_time', 'egu'])

```