BCDA-APS/usaxs-bits #42: How do I pass to plan current RE.md["scan_id"] and link it to epics PV?

jilavsky, 2025-05-26:
Our plans do check occasionally on RE or pull data from RE.md. Somehow, that worked in prior bluesky where RE.... implicitly worked without passing RE into the plan. For example (original Bluesky scans.py):
    bluesky_runengine_running = RE.state != "idle"
...
    SCAN_N = RE.md["scan_id"]+1     # the next scan number (user-controllable)

The new code has RE as optional input parameter for the plan but we do not pass it there. Lines containing RE currently do not compile and RE.md["scan_id"] fails with RE is not defined if I take the RE from the plan parameters or NoneType to which the RE is initialized if the parameter is missing. This looks like some new development in bits I do not understand. 

How do I get the RE.md into plan? 

And how do we create the RE.md["scan_id"] and link it to its epics  value which is usxLAX:specScan ? I have no idea how to approach this in bits. 

Current command line RE.md looks like this:  

In [4]: RE.md
Out[4]: <StoredDict {'beamline_id': '12IDE', 'conda_prefix': '/home/beams/USAXS/.conda/envs/bits_usaxs', 'databroker_catalog': 'usaxs', 'iconfig': {'ICONFIG_VERSION': '2.0.0', 'DATABROKER_CATALOG': 'usaxs', 'RUN_ENGINE': {'DEFAULT_METADATA': {'beamline_id': '12IDE', 'instrument_name': 'APS USAXS/SAXS/WAXS', 'proposal_id': 'Beamline', 'databroker_catalog': 'usaxs'}, 'MD_STORAGE_HANDLER': 'StoredDict', 'MD_PATH': '.re_md_dict.yml', 'USE_PROGRESS_BAR': False}, 'USE_BLUESKY_MAGICS': True, 'BEC': {'BASELINE': True, 'HEADING': True, 'PLOTS': False, 'TABLE': True}, 'NEXUS_DATA_FILES': {'ENABLE': False, 'FILE_EXTENSION': 'hdf'}, 'SPEC_DATA_FILES': {'ENABLE': True, 'FILE_EXTENSION': 'dat'}, 'DM_SETUP_FILE': '/home/dm/etc/dm.setup.sh', 'DEVICES_FILES': ['devices.yml', 'ad_devices.yml', 'shutters_sim.yml'], 'APS_DEVICES_FILES': ['devices_aps_only.yml'], 'MAKE_DEVICES': {'LOG_LEVEL': 'info'}, 'SCALER_PV_NAMES': {'SCALER0_NAME': 'usxLAX:vsc:c0', 'SCALER1_NAME': 'usxLAX:vsc:c1'}, 'USAXS_FLY_SCAN_SETTINGS': {'FALLBACK_DIR': '/share1/USAXS_data', 'SAVE_FLY_DATA_HDF5_DIR': '/share1/USAXS_data/test', 'SAVE_FLY_DATA_HDF5_FILE': 'sfs.h5'}, 'OPHYD': {'CONTROL_LAYER': 'PyEpics', 'TIMEOUTS': {'PV_READ': 5, 'PV_WRITE': 5, 'PV_CONNECTION': 5}}, 'XMODE_DEBUG_LEVEL': 'Plain', 'ICONFIG_PATH': '/home/beams11/USAXS/bits_usaxs/src/usaxs/configs/iconfig.yml', 'INSTRUMENT_PATH': '/home/beams11/USAXS/bits_usaxs/src/usaxs/configs', 'INSTRUMENT_FOLDER': 'configs'}, 'instrument_name': 'APS USAXS/SAXS/WAXS', 'login_id': 'usaxs@usaxscontrol.xray.aps.anl.gov', 'pid': 2196468, 'proposal_id': 'Beamline', 'scan_id': 53, 'versions': {'apsbits': '1.0.3', 'apstools': '1.7.2', 'bluesky': '1.13.1', 'databroker': '1.2.5', 'epics': '3.5.7', 'h5py': '3.13.0', 'intake': '0.6.4', 'matplotlib': '3.10.1', 'numpy': '1.26.4', 'ophyd': '1.10.5', 'pyRestTable': '2020.0.10', 'pysumreg': '1.0.7', 'python': '3.11.12', 'spec2nexus': '2021.2.6'}}>

jilavsky, 2025-05-26:
Ouch, my workaround is wrong type. More fixes needed.

prjemian, 2025-05-26:
> This may be solved, need to test. Closing for now.

In the `iconfig.yml` time, there is a `SCAN_ID_PV` variable for the PV.
Right now, it's commented out.

jilavsky, 2025-05-26:
I uncommented this and connected to usxLAX:specScan, which is epics scan id source. 
It seems to be off as RE.md{"scan_id"] returns 5 while epicsPV has 1. 
Cannot scan, get errors from RE trying to execute scan. 
The best guess is that I am running into problem with epics_scan_id_source function from apsbits/utils/controls_setup. I am getting error that SCAN_ID_SIGNAL_NAME is not defined. Cannot find this anywhere in my code. 

Looking at the old BS, looks like this value (I can only find RUN_ENGINE_SCAN_ID_PV as similar value) should be in iconfig, but when I ook in current iconfig, there is SCAN_ID_PV and no the  SCAN_ID_SIGNAL_NAME.

jilavsky, 2025-05-26:
Cool, this change broke even the lineup2 which worked before. 
As far as I can say, this is the issue. Looks like change to scanID in epics has not been configured as expected? 

```
    new_response = await coro(msg)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1848 in _open_run
    self.md["scan_id"] = self.scan_id_source(self.md)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apsbits/utils/controls_setup.py:44 in epics_scan_id_source
    scan_id_epics = oregistry.find(name=SCAN_ID_SIGNAL_NAME)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/ophydregistry/registry.py:346 in find
    devices = self.findall(
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/ophydregistry/registry.py:504 in findall
    raise ComponentNotFound(
ComponentNotFound: 'Could not find components matching: label="None", name="scan_id_epics"'
```

jilavsky, 2025-05-26:
Commenting the SCAN)ID)PV out restores some of the functionality. 
The code now gets through geometry change, preUSAXStune, measurement of transmission and fails now in usaxs_flyscan.plan. Something I may be able to fix, hopefully.

But this scan_id is here for support group to figure out.

jilavsky, 2025-05-27:
Finally figured this out. Not sure why and what for, but someone (AI?) added to each function parameters following stuff:
```
(...
RE: Optional[Any] = None,
bec: Optional[Any] = None,
specwriter: Optional[Any] = None,
):
```
And since we do not pass these parameters in, they were all set to None. Great, so that means, that the RE.md["scan_id"] cannot exist, since Re is None. 

If there ever was crime in programming against my time, it was adding these parameters for no good reason at all.

closed 2025-05-27 by jilavsky