BCDA-APS/usaxs-bits #44: nexus_flyscan.py reports error related to xml template

jilavsky, 2025-05-26:
After tweaking saveFlyData (/share1/AreaDetectorConfig/Flyscan/saveFlyData.xml) I now have issue with what looks like configuration issue. 

Could be error in template, but I cannot find it. Could be also upgraded support may introduce incompatibility. 

I get KeyError: 'label', but line 325 inb nexus_flyscan.py is inside group specific ation and groups in template do not have labels. 

Confused... 

```

Transient Scan ID: 11     Time: 2025-05-26 12:37:57
Persistent Unique Scan ID: '347a6817-bc10-493b-b7f1-71d5a0c90168'
New stream: 'label_start_motor'
I Mon-12:37:57.305: HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
I Mon-12:37:57.305: HDF5 output: /share1/USAXS_data/2025-05/05_26_test/05_26_test_usaxs/Test_0002.h5
Exception in thread Thread-233 (prepare_HDF5_file):
Traceback (most recent call last):
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/plans/usaxs_fly_scan_plan.py", line 164, in prepare_HDF5_file
    usaxs_flyscan.saveFlyData = SaveFlyScan(fname, config_file=usaxs_flyscan.saveFlyData_config)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/usaxs_flyscan_support/saveFlyData.py", line 93, in __init__
    self._prepare_to_acquire()
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/usaxs_flyscan_support/saveFlyData.py", line 241, in _prepare_to_acquire
    self.mgr._read_configuration()
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/usaxs_flyscan_support/nexus_flyscan.py", line 167, in _read_configuration
    Group_Specification(node, self)
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/usaxs_flyscan_support/nexus_flyscan.py", line 325, in __init__
    self.label = xml_element_node.attrib["label"]
                 ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "src/lxml/etree.pyx", line 2548, in lxml.etree._Attrib.__getitem__
KeyError: 'label'

```

jilavsky, 2025-05-27:
Got it running again. Not sure where out print is ... 
 
```

f.label = xml_element_node.attrib["label"]
                 ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "src/lxml/etree.pyx", line 2548, in lxml.etree._Attrib.__getitem__
KeyError: 'label'
Transient Scan ID: 5     Time: 2025-05-27 10:50:26
Persistent Unique Scan ID: '70645a9c-7302-4d57-a7de-5a74042cb634'
New stream: 'label_start_motor'
I Tue-10:50:26.988: HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
I Tue-10:50:26.988: HDF5 output: /share1/USAXS_data/2025-05/05_27_test/05_27_test_usaxs/Test_0004.h5
E Tue-10:50:26.992: xml_element_node.tag='group' exinfo=KeyError('label')
xml_element_node.tag='group' exinfo=KeyError('label')
Exception in thread Thread-129 (prepare_HDF5_file):
Traceback (most recent call last):
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/usaxs_flyscan_support/nexus_flyscan.py", line 326, in __init__
    self.label = xml_element_node.attrib["label"]
                 ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "src/lxml/etree.pyx", line 2548, in lxml.etree._Attrib.__getitem__
I Tue-10:50:27.006: flying, s    ar, deg      ax, mm       dx, mm       channel      elapsed, s 
KeyError: 'label'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/plans/usaxs_fly_scan_plan.py", line 164, in prepare_HDF5_file
    usaxs_flyscan.saveFlyData = SaveFlyScan(fname, config_file=usaxs_flyscan.saveFlyData_config)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/usaxs_flyscan_support/saveFlyData.py", line 93, in __init__
    self._prepare_to_acquire()
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/usaxs_flyscan_support/saveFlyData.py", line 241, in _prepare_to_acquire
    self.mgr._read_configuration()
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/usaxs_flyscan_support/nexus_flyscan.py", line 167, in _read_configuration
    Group_Specification(node, self)
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/usaxs_flyscan_support/nexus_flyscan.py", line 331, in __init__
    raise RuntimeError(msg)
RuntimeError: xml_element_node.tag='group' exinfo=KeyError('label')

```

jilavsky, 2025-05-27:
Let me try to reduce the xml file to isolate this.

prjemian, 2025-05-27:
These two lines are the logger, then print:

```py
E Tue-10:50:26.992: xml_element_node.tag='group' exinfo=KeyError('label')
xml_element_node.tag='group' exinfo=KeyError('label')
```

Still, need to print out what line in the XML file.  There's some help for that...  somewhere in the https://lxml.de manual.

prjemian, 2025-05-27:
I will fix this in the next couple days.

PR #50 (2025-05-28): Restore correct XML parser code to nexus flyscan writer
- close #44

@jilavsky This PR is in _draft_ mode.  It should be tested before merging.