BCDA-APS/usaxs-bits #46: Specwriter error

jilavsky, 2025-05-27:
I though we fixed specwriter by upgrading...

This comes at the end for Flyscan. It was not there with previous version of specwriter. That used to work. 

```
xceptions = self.cb_registry.process(name, name.name, doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:422 in process
    func(*args, **kwargs)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:512 in __call__
    return mtd(*args, **kwargs)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/callback_base.py:95 in receiver
    handler(doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/spec_file_writer.py:812 in event
    self.write_scan_data_row(doc)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/callbacks/spec_file_writer.py:992 in write_scan_data_row
    for label in self.data_labels:
AttributeError: 'SpecWriterCallback2' object has no attribute 'data_labels'
```

prjemian, 2025-05-27:
See https://github.com/BCDA-APS/apstools/issues/1059

prjemian, 2025-05-27:
Hasn't been released in public apstools version.

jilavsky, 2025-05-27:
Works

commit 02b3a7f8 (2025-05-27): Fixes  #46 and few other problems
closed 2025-05-27 by jilavsky