BCDA-APS/usaxs-bits #59: Name transmission measurements

jilavsky, 2025-05-30:
Transmission plans currently do not set plan name and have empty name in BITS listruns().

prjemian, 2025-05-30:
If it did not generate a run, it would not show up as an unnamed run, right?

prjemian, 2025-05-30:
Here is where a run is created: https://github.com/BCDA-APS/usaxs-bits/blob/cd24b41111b09388ffaf33d534c7801a4efdbe79/src/usaxs/plans/sample_transmission.py#L309

prjemian, 2025-05-30:
Here's the problem: https://github.com/BCDA-APS/usaxs-bits/blob/53df6aa269ebef8c8a2e9139de4993ec57760e23/src/usaxs/plans/filter_plans.py#L129

It's a copy/paste typo.  Here's the old code: https://github.com/BCDA-APS/usaxs-bits/blob/62f74b673bc14a76fe1085fe350c60de613f9291/instrument/plans/filters.py#L57-L62

prjemian, 2025-05-30:
Tested to work as expected at the beam line.