BCDA-APS/usaxs-bits #60: Fix baseline items missing value_start

jilavsky, 2025-05-30:
These errors are reported now:
softioc with sample data runs and PVs exist. Check to be sure. 
undulator should still be there ;-) 

```
plan uascan ['6e81bd44'] (scan num: 10)
W Thu-19:21:45.111: 'HDF5 address /entry/instrument/bluesky/streams/baseline/undulator_downstream_device/value_start not found.' -- not creating undulator group
W Thu-19:21:45.114: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_chemical_formula/value_start not found.'
W Thu-19:21:45.114: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_concentration/value_start not found.'
W Thu-19:21:45.115: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_description/value_start not found.'
W Thu-19:21:45.115: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_electric_field/value_start not found.'
W Thu-19:21:45.115: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_magnetic_field/value_start not found.'
W Thu-19:21:45.115: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_rotation_angle/value_start not found.'
W Thu-19:21:45.115: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_scattering_length_density/value_start not found.'
W Thu-19:21:45.116: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_stress_field/value_start not found.'
W Thu-19:21:45.116: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_temperature/value_start not found.'
W Thu-19:21:45.116: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_volume_fraction/value_start not found.'
W Thu-19:21:45.116: 'HDF5 address /entry/instrument/bluesky/streams/baseline/sample_data_x_translation/value_start not found.'
W Thu-19:21:45.116: no sample data found, not creating sample group
W Thu-19:21:45.117: 'HDF5 address /entry/instrument/bluesky/streams/baseline/bss_user_info_contact/value_start not found.' -- not creating source group
Exception in thread Thread-2017 (_threaded_writer):
Traceback (most recent call last):
```

prjemian, 2025-05-30:
Undulator defined previously: https://github.com/BCDA-APS/usaxs-bits/blob/62f74b673bc14a76fe1085fe350c60de613f9291/instrument/devices/aps_undulator.py#L17C1-L18C1

adding now:

```yaml
apstools.devices.PlanarUndulator:
- name: undulator
  prefix: "S12ID:USID:"
  labels: 
  - baseline
```

prjemian, 2025-05-30:
Ophyd device named `sample_data` defined here: https://github.com/BCDA-APS/usaxs-bits/blob/6fed7bd271f6f94f5a56b8ef4c45ac036d5226ba/src/usaxs/devices/sample_data.py#L97

New BITS configuration creates `sample_data_device` which includes `baseline` label: https://github.com/BCDA-APS/usaxs-bits/blob/6fed7bd271f6f94f5a56b8ef4c45ac036d5226ba/src/usaxs/configs/devices.yml#L103-L106

Will remove the old style object created in `devices/` and rename `sample_data_device` to `sample_data`.

prjemian, 2025-05-30:
Commit 53df6aa269ebef is for this issue.

closed 2025-05-30 by prjemian