BCDA-APS/usaxs-bits #63: BITS fails to collect AreaDetector images

jilavsky, 2025-06-02:
Occasionally, twice in the last two days, BITS lost ability to collect images. Once with BlackFly Optical and one with usaxs_Eiger1 (WAXS) detector. Recorded error with WAXS detector. Looks like BITS timed out on put to filename. Note that the timeout was 60 seconds!

Softioc worked fine from epics screen. After softioc restart BITS collected data (without BITS restart) without any issues. 
Suspect network issues? Hopefully related to upcoming APS startup and finalizing some network changes? 

```
Transient Scan ID: 5     Time: 2025-06-01 13:27:01
Persistent Unique Scan ID: '84990f08-dac1-42ff-98b0-2ae4ba999a33'
I Sun-13:27:02.894: Finished SAXS/WAXS data collection in 2.34 seconds.
I Sun-13:27:02.938: Collected SAXS with HDF5 file: /share1/USAXS_data/2025-05/06_01_Eric/06_01_Eric_saxs/test_0036.hdf
I Sun-13:27:02.950: Starting collection of WAXS for test
I Sun-13:27:02.967: Moving to WAXS mode ... please wait ...
I Sun-13:27:09.568: Creating technique directory: /share1/USAXS_data/2025-05/06_01_Eric/06_01_Eric_waxs
W Sun-13:28:09.716: waxs_det_hdf1_file_name: _set_and_wait(value=test, timeout=60, atol=None, rtol=None, kwargs={})
E Sun-13:28:09.716: Exception FailedStatus during attempt 1 of 1 of command '('waxsExp', ['-140', '20', '1', 'test'], 27, '      waxsExp        -140      20      1      "test"')''
subject: FailedStatus during attempt 1 of 1 of command '('waxsExp', ['-140', '20', '1', 'test'], 27, '      waxsExp        -140      20      1      "test"')''

date: 2025-06-01 13:28:09.716916
command file: /share1/USAXS_data/2025-05/usaxs.mac
line number: 27
command: ('waxsExp', ['-140', '20', '1', 'test'], 27, '      waxsExp        -140      20      1      "test"')
raw command:       waxsExp        -140      20      1      "test"
attempt: 1 of 1
exception: Status(obj=EpicsSignalWithRBV(read_pv='usaxs_eiger1:HDF1:FileName_RBV', name='waxs_det_hdf1_file_name', parent='waxs_det_hdf1', value='t', timestamp=1748802429.636288, auto_monitor=True, string=True, write_pv='usaxs_eiger1:HDF1:FileName', limits=False, put_complete=False), done=True, success=False)
Stopping further processing of this command list.

Out[2]: 
('5e88265e-b9af-4509-806f-aa1a00650a4a',
 '0ec1f7d4-97dd-4774-ade7-e4b98ff0a13e',
 'e1d8cc76-845a-4334-a6f4-52af113fb62c',
 '84990f08-dac1-42ff-98b0-2ae4ba999a33')
```

jilavsky, 2025-06-03:
Just to make sure we have all background, this is the part of code which sets the PV which failed in above code
```py
    WAXSscan_path = techniqueSubdirectory("waxs")
    WAXS_file_name = local_file_template % (
        scan_title_clean,
        waxs_det.hdf1.file_number.get(),
    )
    _md["hdf5_path"] = str(WAXSscan_path)
    _md["hdf5_file"] = str(WAXS_file_name)

    pilatus_path = os.path.join("/mnt/share1", *WAXSscan_path.split(os.path.sep)[2:])
    if not pilatus_path.endswith("/"):
        pilatus_path += "/"
    local_name = os.path.join(WAXSscan_path, WAXS_file_name)
    logger.debug(f"WAXS HDF5 file: {local_name}")
    pilatus_name = os.path.join(pilatus_path, WAXS_file_name)
    logger.debug(f"Pilatus computer Area Detector HDF5 file: {pilatus_name}")

    waxs_det.hdf1.file_path._auto_monitor = False
    waxs_det.hdf1.file_template._auto_monitor = False
    yield from bps.mv(
        # fmt: off
        waxs_det.hdf1.file_name,         scan_title_clean,
        waxs_det.hdf1.file_path,         pilatus_path,
        waxs_det.hdf1.file_template,    ad_file_template,
        timeout=MASTER_TIMEOUT,
        # fmt: on
    )
    waxs_det.hdf1.file_path._auto_monitor = True
    waxs_det.hdf1.file_template._auto_monitor = True
    # paths and names done by now
```

jilavsky, 2025-06-03:
Note, that there is no RBV value for this field. How is this working at all? 

```

(base) usaxs@usaxscontrol /share1/USAXS_data $ caget usaxs_eiger1:HDF1:FileName.RBV
Channel connect timed out: 'usaxs_eiger1:HDF1:FileName.RBV' not found.
(base) usaxs@usaxscontrol /share1/USAXS_data $ caget usaxs_eiger1:HDF1:FileName
usaxs_eiger1:HDF1:FileName 256 116 101 115 116 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
(base) usaxs@usaxscontrol /share1/USAXS_data $

```

jilavsky, 2025-06-03:
As far as I can say, this should be EpicsSignal and not ExpicsSignalWIth RBV

jilavsky, 2025-06-03:
This works:
```
(base) usaxs@usaxscontrol /share1/USAXS_data $ caget usaxs_eiger1:HDF1:FileName_RBV
usaxs_eiger1:HDF1:FileName_RBV 256 116 101 115 116 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
(base) usaxs@usaxscontrol /share1/USAXS_data $
```

prjemian, 2025-06-03:
It's an area detector pattern. These PVs do not have RBV fields. Instead, they use a second PV with `_RBV` instead of `.RBV`