BCDA-APS/usaxs-bits #69: Re.abort drops bits bs in broken plot state

jilavsky, 2025-06-10:
Users did ctrl-c and then RE.abort(). That all worked fine, but after that bits-bs was in weird state where bec plots did not work. Lack of bec plot updates caused user concern and assumption something is wrong. Fix was to restart bits-bs. 
Need to find way to properly reset usaxs using function to working state. Need to revise resetUSAXS code.

ravescovi, 2025-06-11:
> Assigned #69 <https://github.com/BCDA-APS/usaxs-bits/issues/69> to
> @ravescovi <https://github.com/ravescovi>.

I am a bit confused. They used `ctrl+C` once. It killed the RE. Then they
did RE.abort()?
Why is restarting the whole bits-bs (terrible name hehehe) bad?

jilavsky, 2025-06-16:
Since this is end of run for now, I cannot test if bits still work or if restart is required.

jilavsky, 2025-07-20:
Did not seem to be issue anymore.

closed 2025-07-20 by jilavsky