BCDA-APS/usaxs-bits #72: Suspender crashed

jilavsky, 2025-07-10:
I was wrong in previous testing due to various other interruptions. 

This time I had a chance to test cleanly our current suspenders and it resulted in crash of BS, not proper restart...

```
I Thu-13:26:44.838: HDF5 file complete: /share1/USAXS_data/2025-07/07_09_Larry/07_09_Larry_usaxs/BoehNaCl6mLE_26_0665.h5
I Thu-13:26:46.089: Starting collection of SAXS for BlankLE_26
I Thu-13:26:46.115: Moving to SAXS mode ... please wait ...
Suspending....To get prompt hit Ctrl-C twice to pause.
Suspension occurred at 2025-07-10 13:27:00.
Justification for this suspension:
Signal usaxs_CheckBeamStandard is low
Suspending....To get prompt hit Ctrl-C twice to pause.
Suspension occurred at 2025-07-10 13:27:32.
Justification for this suspension:
Signal FE_shutter_pss_state = 0 fell below 1 and has not yet crossed above 1.
Suspender SuspendFloor(EpicsSignalRO(read_pv='PA:12ID:STA_A_FES_OPEN_PL', name='FE_shutter_pss_state', parent='FE_shutter', value=1, timestamp=1752174222.733616, auto_monitor=True, string=False), sleep=0, pre_plan=None, post_plan=None, tripped_message=) reports a return to nominal conditions. Will sleep for 0 seconds and then release suspension at 2025-07-10 14:03:42.
E Thu-14:03:42.759: Error in measure_SAXS_Transmission: Cannot bundle readings without an open run. That is, 'create' must be preceded by 'open_run'.
Suspender SuspendBoolLow(EpicsSignalRO(read_pv='usxLAX:blCalc:userCalc1', name='usaxs_CheckBeamStandard', value=0.0, timestamp=1752172020.618668, auto_monitor=False, string=False), sleep=0, pre_plan=None, post_plan=None, tripped_message=) reports a return to nominal conditions. Will sleep for 0 seconds and then release suspension at 2025-07-10 14:03:42.
E Thu-14:03:42.759: Run aborted
Traceback (most recent call last):
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py", line 1609, in _run
    msg = self._plan_stack[-1].send(resp)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/apstools/plans/labels_to_streams.py", line 231, in label_stream_wrapper
    return (yield from bpp.plan_mutator(plan, action[when.value]))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 181, in plan_mutator
    raise ex
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 134, in plan_mutator
    msg = plan_stack[-1].send(ret)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 1375, in __call__
    return (yield from plan)
            ^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 1228, in baseline_wrapper
    return (yield from plan_mutator(plan, insert_baseline))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 181, in plan_mutator
    raise ex
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 134, in plan_mutator
    msg = plan_stack[-1].send(ret)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 845, in monitor_during_wrapper
    return (yield from plan2)
            ^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 181, in plan_mutator
    raise ex
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 134, in plan_mutator
    msg = plan_stack[-1].send(ret)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 181, in plan_mutator
    raise ex
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 134, in plan_mutator
    msg = plan_stack[-1].send(ret)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 907, in fly_during_wrapper
    return (yield from plan2)
            ^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 181, in plan_mutator
    raise ex
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 134, in plan_mutator
    msg = plan_stack[-1].send(ret)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 181, in plan_mutator
    raise ex
  File "/home/beams/USAXS/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py", line 134, in plan_mutator
    msg = plan_stack[-1].send(ret)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beams11/USAXS/bits_usaxs/src/usaxs/user/finite_loop.py", line 242, in myFiniteListLoop
[... 171 lines cut ...]
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py:206 in plan_mutator
    inner_ret = yield msg
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py:206 in plan_mutator
    inner_ret = yield msg
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py:206 in plan_mutator
    inner_ret = yield msg
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/preprocessors.py:206 in plan_mutator
    inner_ret = yield msg
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1584 in _run
    msg = self._plan_stack[-1].throw(stashed_exception or resp)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:192 in single_gen
    return (yield msg)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1584 in _run
    msg = self._plan_stack[-1].throw(stashed_exception or resp)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1299 in suspender_helper_inner_plan
    yield from rewind_plan
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/utils/__init__.py:171 in <genexpr>
    gen = (msg for msg in gen)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1676 in _run
    new_response = await coro(msg)
  File ~/.conda/envs/bits_usaxs/lib/python3.11/site-packages/bluesky/run_engine.py:1938 in _create
    raise IllegalMessageSequence(ims_msg)
IllegalMessageSequence: Cannot bundle readings without an open run. That is, 'create' must be preceded by 'open_run'.

```

jilavsky, 2025-07-11:
This line in error message:

E Thu-14:03:42.759: Error in measure_SAXS_Transmission: Cannot bundle readings without an open run. That is, 'create' must be preceded by 'open_run'.

makes me wonder if the first case resulting in crash was suspended in wrong time. 

Second time I observed suspending and it happened in the middle of Flyscan. And recovery was perfect. 

Does measure_SAXS_transmission need to be written in different way? Is it suspend-able plan? Looks to me it did not have open_run, whatever that means.

prjemian, 2025-07-11:
Line 45 of https://github.com/BCDA-APS/usaxs-bits/blob/main/src/usaxs/plans/sample_transmission.py says that this plan should not be a bluesky run. Where is it calling create()?  That's a mistake.

jilavsky, 2025-07-20:
Added proper decorator. For now seems to be working, but we may have not failed at the right moment. May be test during commissioning somehow?