BCDA-APS/usaxs-bits #76: Add structure to user folders created by newUser()

jilavsky, 2025-07-22:
User request - as well as preparation for potential future use of Globus for data distribution...

Users would like to add structure to currently used user folder created by newUser command. Since we have users who run multiple dissimilar experiments, including different setups and even different X-ray energies, they want to keep data separate. Also, Matilda could get confused by one folder with multiple blanks if they do not belong together. 

To make data sharing and management easier, we should enable more complex structure:
current: 
2025-07 : 07_15_Fan : 07_15_Fan_usaxs
                                       07_15_Fan_saxs
                                       07_15_Fan_waxs

Future:
2025-07 : 07_15_Fan : Sample1 : Sample1_usaxs
                                                      : Sample1_saxs
                                                      : Sample1_waxs
                                     : Sample2 : Sample2_usaxs
etc... 

Questions: 
do we need to force new structure on all or can we use both? 
Can we modify newUser() or do we need newSample()? Should newUser call newSample() to make sure we get proper structure?  
How does this impact data saving code for Flyscan, Stepscan and for all involved area detectors? 
Any impact on Matilda or Igor ? 
What other ugly bugs this will provoke?

prjemian, 2025-07-22:
The APS Data Management folder structure is:

```
EXPERIMENT_NAME/
  analysis/
    # any folder/files needed by or produced by data handling, reduction, analysis, visualization, ...
    # You get to decide the structure under `analysis/`
  data/
    # raw data folders/files
    # You get to decide the structure under `data/`, including whatever you consider "raw" data.
  system/
    # Don't you add nothing here, leave it for the data management system admins
```

EXPERIMENT_NAME is the unique APS Data Management "experiment" you create for this data.  This associates an experiment name with file storage, users with varying access permissions, proposal, ESAF, and beam time allocations.

banubot, 2025-07-22:
You can use both. When creating the experiment, set the storage root path to 2025-07/07_15_Fan/Sample1 or 2025-07/07_15_Fan and the experiment name to 07_15_Fan_usaxs or Sample1_usaxs. Experiment names must be unique for your station, so 07_15_Fan_Sample1_usaxs might be a better experiment name even if it is redundant